supermerill/SuperSlicer #784: Layer height not used correctly

BelgianGuyWithALaser, 2020-12-09:
Version
2.2.53

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
CR20PRO (not stock) klipper firmware

### Behavior
layer heigth not used as Filled inside the settings.
Set nozzle to 0.4mm in printer settings, set first layer heigth to 0.3 (or 75% with both versions it's not working) slice and check layer heigth in the preview. it becomes 0.32 instead of the asked 0.30.

expected layer heigth of 0.30, 
Actual Results received 0.32

  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]

#### Project File (.3MF) where problem occurs
[layerheigtherror.zip](https://github.com/supermerill/SuperSlicer/files/5667667/layerheigtherror.zip)

supermerill, 2020-12-09:
what are you z full step ?

supermerill, 2020-12-09:
You have your z_step at 0,04: so your layer height can be 0.28 or 0.32

BelgianGuyWithALaser, 2020-12-09:
even when setting the Z_step to 0 both it's still at 0.32

supermerill, 2020-12-09:
That's a refresh bug. Set the first layer hieght to something else then back to 75%

closed 2021-06-13 by supermerill