supermerill/SuperSlicer #785: FR Settings Sync

ReubenBTalbott, 2020-12-10:
Feature Request Settings Sync.

I don't know how easy this would be to add or if you would even be interested in adding it but here it is.

I use multiple computers and I was thinking it would be amazing to have a way to sync settings easily between devices, I then thought about the VS Code plugin [Settings Sync ](https://marketplace.visualstudio.com/items?itemName=Shan.code-settings-sync) which syncs  VScode settings and plugins using a GitHub Gist after you sign into your GitHub account. Would it be possible to add a similar feature to SuperSlicer? I realize that a Gist probably won't work but perhaps SuperSlicer could use a Private Repository?

Thanks for making my favorite slicer PrusaSlicer even better!
Reuben

ProfEngr, 2020-12-16:
It would definitely be better than export/import that I have to do now between the PCs where I use SS. Maybe the ability to specify a shared (Drive, Box, etc) file location for the configs?

neophyl, 2020-12-16:
If you start Susie with the --datadir switch you can specify the profiles folder location.  You can then use something like dropbox to store the profiles and then its synced across multiple computers.  Doesn't that do what you want already ?

ReubenBTalbott, 2021-01-08:
> If you start Susie with the --datadir switch you can specify the profiles folder location.  You can then use something like dropbox to store the profiles and then its synced across multiple computers.  Doesn't that do what you want already ?

That would definitely help, and I'll see if I can set that up, but I still think it would be useful for it to be integrated directly into SuperSlicer. One reason for me is that my only cloud storage is iCloud which only supports MacOS and Windows and some of my machines are running Linux which has no iCloud client.

neophyl, 2021-01-08:
The other way you can do it is to copy the profile folders and the folder for Susie to a USB stick and point at them.  That way you can run it entirely portable from the stick.  I know that works as I did that myself for awhile.  You can then have whatever versions of Susie you want on the stick and just run the one you want.  Its what I do still when I want to use different versions (like 2.2.54 as well as 2.2.55 for example).

supermerill, 2021-06-02:
I'm going to add a sync option, but not now.

kklimczak, 2021-06-11:
Are profile files a plain text? This will be perfect because files can be stored in the version control system like a git.

neophyl, 2021-06-11:
If you look in your configuration folder you will see that they are stored as .ini files and as such are plain text yes.
The Vendor files are also of a plain text type also but appear to use a slightly different format with headers etc so that the dependencies all work.

johnmeacham, 2021-07-13:
I use unison https://www.cis.upenn.edu/~bcpierce/unison/ to synchronize my superslicer configs (along with all my configs and active projects) between computers and it works great.

Gobol, 2021-09-16:
MacOS 11.2.3, Susie 2.3.57, specifying --datadir to network location mounted under /Volumes/ makes Susie not running at all without giving a msg. Nothing in Console, either. 
Symlinking ~/Library/Application Support/SuperSlicer to network location (available) makes Susie crash after open with error 255. (also in Console).
On Win64 platform it works ok.

supermerill, 2021-09-16:
@Gobol is it a bug specific on 2.3.57 that doesn't trigger on 2.3.56 ?

Gobol, 2021-09-18:
@supermerill Checked right now - 2.3.56 doesn't work the same way as 2.3.57 - it means, after passing --datadir argument to ./SuperSlicer - nothing is happening. No bouncing icon on dock, nothing. No sign of some error in Console.app too (system.log). No crash report either.

Gobol, 2021-09-18:
@supermerill And of course, I've checked permissions to run (sec & priv page). 
running `./SuperSlicer` without args works correctly.

supermerill, 2021-09-27:
I check the diff and I don't see why there is a difference between the versions...
Does the bug also appear in prusaslicer 2.3.3 vs 2.3.0 ?
I don't know how macos "install" the .dmg, does it extract it somewhere?

Gobol, 2021-09-27:
Meh, my bad. Sorry, didn't make myself clear - not a native english writer...
Both versions of SuperSlicer behave the same way - didn't start application when specified --datadir argument.
Didn't checked PrusaSlicer, though. I'll try soon.

MacOS installation is simply extraction/copy from DMG to /Applications directory on system disk, unless there are some specific pre/post-install scripts, but that's beyond my scope of cognition... sorry.

I'll try `--loglevel 5` and drop a line with results.

Gobol, 2021-09-27:
Ok, the log:

`gobo@Gobos-MBP MacOS % ./SuperSlicer --loglevel 5 --datadir /Volumes/NAS/SuperSlicer
[2021-09-27 22:05:29.783979] [0x000000011425ae00] [debug]   full path: 15840400373692917632
[2021-09-27 22:05:29.784614] [0x000000011425ae00] [debug]   single instance: undefined. other params: ./SuperSlicer;--loglevel;5;--datadir;/Volumes/NAS/SuperSlicer
[2021-09-27 22:05:29.784647] [0x000000011425ae00] [debug]   full lock path: /Volumes/NAS/SuperSlicer/cache/15840400373692917632.lock
[2021-09-27 22:05:29.807877] [0x000000011425ae00] [info]    instance check: Another instance found. This instance will terminate.`

Problem is - there is no `.lock` file before exec line in `datadir/cache`.
SuperSlicer creates one after start, and drops error that it found another instance (seemingly self one)...

Bupher, 2022-01-07:
Not to drag up a dead thread, but I have a workaround for this. I use it for prusa slicer with no problems. I make a symbolic link from my synced documents folder to appdata/roaming and rename the link SuperSlicer. This has worked in prusa slicer for over a year, and seems to be working well for me in SuperSlicer as well.

This was on windows, so I use the link shell extension and do it all from the right click context menu. I assume on link you could use something like Rsync or similar.

supermerill, 2022-01-07:
now if you have a directory named "configuration" in your superslicer 'install' dir, then it will be used instead of the appdata/roaming.

Still working on a proper sync thing, but I haven't much time to dedicate to it.

OldCurmudgeon3DP, 2022-01-08:
I use a dropbox location and point the BAT file there. I just have to remember to open the program shortcut 1st instead of opening the file via OLE.

Misterke, 2022-01-25:
I actually checked in my appdata/Roaming/SuperSlicer folder into a git repo.  That way, when I have done a useful change to one of the profiles on one of my PCs, I commit that, push it to the central git server (I run gitea) and can then fetch that on other PCs and merge in the changes.  I have to merge them in, because even just starting up SuperSlicer changes some of the .ini files and hence generates conflicts.  Most of the time those are very easy to fix, but it does take additional manual actions.  It would be great if every save of a profile could have SuperSlicer automatically create a commit and push that to the server and then on other systems "pull" that change in.  Having it all in git has the super benefit of having all history at your fingertips...  And to be honest, even when manually using git in this way with SuperSlicer, it still beats Cura's sync to the cloud.

supermerill, 2022-04-11:
if you add the `--datadir c:/my/local/gsync/susi/conf/dir` parameter to your shortcut, you should be good.

I still plan to add a built-in sync, but I don't have the time for that yet.