supermerill/SuperSlicer #794: Fan step-up and step-down code issuing too many extra speed commands

HaythamB, 2020-12-11:
### Version
2.3.55.2

### Operating system type + version
Windows 10 Home (64-bit)

### 3D printer brand / version + firmware version (if known)
Custom toolchanger running RRF2 on a Duet2 + Duex 5

### Behavior
* While utilizing the "Run the fan at default speed" setting with a min of 50% and a max of 100%, the sliced gcode file has inserted a large repetition of changing the fan speed between 100% and some other calculated value. This leads to 50 sequential fan speed gcodes which causes the print to stutter while executing the un-needed fan speed adjustments.

Tested same settings in stable 2.2 and Prusaslicer alpha (PrusaSlicer-2.3.0-beta2+win64-202012051115) and the issue isn't occurring in either of those releases.

* Set Filament Settings -> Cooling to: "Run fan at default speed when possible" and set minimum and maximum to 2 different numbers. Check exported gcode

* Simple ramp-up and step-down fan commands based on perimeter, bridge, etc, that is outputted once as normal.
* Up to 50x M106 gcode commands in sequence (see line 1821 and below in the attached gcode output)
[image]
[image]

#### Project File (.3MF) where problem occurs

[SS Fan Issue - Hemera Volcano fan duct.zip](https://github.com/supermerill/SuperSlicer/files/5678000/SS.Fan.Issue.-.Hemera.Volcano.fan.duct.zip)

ardyesp, 2020-12-12:
I am getting a virus warning when downloading your zip file!!

HaythamB, 2020-12-12:
I think something might be wrong on your machine..

[image]

ardyesp, 2020-12-13:
Yeah, the warning was from Firefox which does not like it for some reason. Chrome worked ok.

Whitehawk2000, 2020-12-15:
Maybe your issue is somehow related to the one i have. It always stutters when ramping the fan up before overhangs.
[https://github.com/supermerill/SuperSlicer/issues/689](url)

MattAtA3D, 2022-12-21:
So our issue isn't _quite_ as severe as the OPs, but are there options to tame this? We're getting segments like this all over the place:
```gcode
M106 S76.5
;WIDTH:0.472969
M106 S153
G1 X529.68 Y340.642 E2.24456
M106 S76.5
;WIDTH:0.569546
M106 S153
G1 X529.593 Y340.709 E2.25117
M106 S76.5
;WIDTH:0.536678
M106 S153
G1 X529.642 Y340.831 E2.25854
G1 X529.9 Y341.438 E2.29567
G1 X530.158 Y342.046 E2.3328
G1 X530.263 Y342.293 E2.34789
G1 X529.769 Y342.503 E2.37807
G1 X529.664 Y342.256 E2.39316
G1 X529.148 Y341.04 E2.46741
M106 S76.5
;WIDTH:0.569241
M106 S153
G1 X529.095 Y340.911 E2.4758
G1 X528.962 Y340.913 E2.48379
M106 S76.5
;WIDTH:0.474154
M106 S153
G1 X528.829 Y340.916 E2.49031
G1 X528.593 Y341.008 E2.50273
G1 X528.356 Y341.1 E2.51516
G1 X528.231 Y341.184 E2.52252
M106 S76.5
;WIDTH:0.576349
M106 S153
G1 X528.106 Y341.267 E2.53168
G1 X528.117 Y341.459 E2.5434
G1 X528.908 Y343.321 E2.66645
M106 S76.5
M106 S153
G1 X529.091 Y343.752
G1 X529.115 Y343.786
M106 S76.5
;WIPE_START
;WIPE_END
```

These flares don't do us any good - our fan's inertia and momentum are large enough that it doesn't even pretend to notice them.

In a few specific instances, these splats of fan changes have caused issues with our printer's software (customized from a 3rd party, we cannot change it easily in the best cases). So pointers to any existing knobs we can turn to reduce them would be greatly appreciated.