supermerill/SuperSlicer #795: Feature Request: "brim" for draft shield

jhoblitt, 2020-12-12:
I'm currently printing with a filament that benefits from using a draft shield.  As this material likes to warp, a single perimeter draft shield tends to warp and detach from the bed.  It would be really useful if there was an option to add a brim to the outside of the draft shield.

Currently, if you want a "skinny" draft shield without using a bunch of "loops", the only way to achieve it is to add a brim to the main part such that the draft shield overlaps part or all of the brim.

chron0, 2020-12-14:
yes! I would love this too. In the past I always had to build the draft-shield (called it cocoon then) in the model as a one perimeter width object and had to align n-loops of brim right below it to make it stick. Works, but is cumbersome. If we only would have two different options for draft shield loops (1) and brim loops (3-5) to set separately it would be a total win. draft shield should either begin on the innermost loop or also be user selectable.

EvilGremlin, 2021-02-09:
Vote, long long awaited feature, i thought it was obvious thing to do from the start?

supermerill, 2021-06-02:
You can do that, using a brim width > skirt distance and brim offset < skirt distance.

I understand it's a bit cumbersome.
A skirt brim can be a `loops (min) for the first layer`. should the next layers be centred or on the object side of the 'brim'?

EvilGremlin, 2021-06-02:
This is exactly how i do that, but it's very bad option. Because brim is equidistant and skirt is not.