supermerill/SuperSlicer #796: Perimeter bulge issue

cylstudio, 2020-12-12:
### Version
2.2.53

### Operating system type + version
Windows 10 Pro 64 bit (10.0.18363 Build 18363)

### 3D printer brand / version + firmware version (if known)
Prusa Mini (latest version)

### Behavior
* _Describe the problem_
Encountered issue with the outside wall, keep getting this strange Bulge around the area when solid meet the walls.  I have tried all different settings with the Perimeter...no luck.
* _Steps needed to reproduce the problem_
Easily reproduce, just print the STL file I've provided.
* _Expected Results_
Smooth outside wall
* _Actual Results_
Strage bulge line, seems to be a very common issue with PrusaSlicer, and I thought I give SuperSlicer a go, same results.

#### Project File (.3MF) where problem occurs
[Bulge_test_3mf.zip](https://github.com/supermerill/SuperSlicer/files/5683103/Bulge_test_3mf.zip)

[image]
Two test on the left were sliced by CURA, seems pretty clean...and all of the rest on the right sliced by PrusaSlicer and SuperSlicer

mucomuco, 2020-12-12:
I have exact same problem. It only happens with SUSI and PRUSAslicer.  This problem doesnt happen if it is a simple shape box but when there is ledge on walls these bulges are happening just starting of that ledge. 
On this pring u can see. It happens on starting of hole and starting point of any kind of ledge.

[image]

[image]

wildjokerss, 2020-12-12:
@cylstudio I took a peek out of curiosity on your 3MF and noticed that the area of that "bulge" seems to match the area where your "min layer time" is kicking in, and instead of printing at ~30mm/s it is printing at 18. Perhaps worth checking that out. Screenshot for reference.
[image]
[image]

you can try to decrease it or disable it so everything is done at the same speed and see if that weird shenanigans keep showing up

(bear in mind that I'm just a curious / interested guy trying to help, not the guru here) :)

cylstudio, 2020-12-13:
Hi there, thanks for your contribution to the problem, that's a very good thought however still no luck.
I have disabled it the layer time goal so that all outer perimeter print at the same speed, results is identical as below, very very strange. Do you mind to print the same STL using SuperSlicer?

[image]
[image]

[image]
[image]

wildjokerss, 2020-12-13:
This pics are making it way more clear. Indeed there is some weird shenanigans going on there. I myslef found a issue on a benchy that seems to "break" the perimeters flow or something and it looks weird. Check below:
[image]
In only affects those perimeters and although this is a stopped print, the rest of the benchy is fine. 
I'll try yours at least with your print profile and let you know once I do it

wildjokerss, 2020-12-13:
@cylstudio I'm back :D
So, although there seems to be clearly some weird shenanigan going on that part of the model, mine does not seem as bad as yours. This was with your print profile, only changed to my filament and to my printer and printed it.
[image]
[image]

neophyl, 2020-12-13:
Have any of you read https://help.prusa3d.com/en/article/the-benchy-hull-line_124745  ?

wildjokerss, 2020-12-13:
Hi @neophyl , I actually noticed that my benchy seemd to have a few issues, however, the very same STL on ideamaker worked very well, which is what made me think that it was a beef with the slicer. Take a look below, it's the same filament, profile and printer, only sliced with ideamaker:
[image]
However, you're right, that article seems to point out to a fixed bency that I'm gonna take a shot now just to confirm it was an STL issue, so I can apologise to my printer for all the names I've called her 😂 Thanks for sharing that @neophyl !

neophyl, 2020-12-13:
Sorry, I don't mean to imply that the slicer couldn't possibly do better.  It possibly/probably could.  Some slicers do seem to handle it better but ALL of them do suffer from these issues, just to different extents and it differs from model to model.

The point being that this is an 'evergreen' issue.  Its been around forever, years and years and has so many issues and forum threads about it.  Its part of Slic3er, Prusa Slicer, SuSie etc and lots of others that I don't use most likely.  I don't think its going to be a simple 'fix it this way'.   If it was it would already have been fixed.  The Prusa article is mainly interesting as it touches on why its such a difficult issue and that its effected by so many different factors.

wildjokerss, 2020-12-13:
yeah they sure do, I didn't took your wording like you were implying something :) 
I've although checked the STL on the page that you shared and the slicer seems to be doing the same, it also has the same number of issues so I'm guessing that there could be a problem with the STL's we using. but let's be honest, if the STL is broken or somewhat damaged we can't put the blame on the slicer either, thing's working with what it got :P
Time for a new benchmark thingy :D

mucomuco, 2020-12-13:
I have been trying to fix this issue for weeks now. I spent one filament spool on it cuze of doing test prints. I tried this on simply and  it doesnt have this problem. As i said above if you do a simple box and infill it then there is no problem at all. My printer prints it perfectly with SUSI or prusaslcier but when there is layer shifts, not a simple box obeject then this bulges are appearing. It is like there is not enough space for filament and it is overflowing . Speed is not making the problem at first place it only make it worse. I am not a pro but as i can see it is a slicing problem. btw It can be less visible if you play with overlap options. I wasted so much time on this thing and it is annoying, it is annoying that dont know what is causing it. It is not only happing on one stl file it happens on every print which has layer shifts. It can be minimizing by slicers setting probably but "print it on low speed" isnt a solution at all. I printed on simply with 50mm/s full speed on all perimeters and it printed without any problem. I am priting on 15mm/s on SUSI it has same problem. Btw if you disable anchored/connected option in infill options it is printing without any problem. But i am not sure if it is good for printing cuze there is no connection with infill and walls. Infill lines only touchs them with that option disabled. Really annoying and confusing thing :((

CCS86, 2020-12-14:
Sounds like a good opportunity for an experimental feature to use variable print speed modifiers to normalize total layer time, or at least make the changes more gradual. 

Since it is maybe not practical to speed up larger layers, you could just slow down smaller layers. Maybe you could define a maximum change in layer time (either in % or seconds), and bounded by minimum print speed, allow the slicer to slow layers down to meet this maximum layer time delta.

supermerill, 2021-06-02:
I tried to reproduce but I couldn't. I can try again if someone has a good profile of a simple object (on ender3 if possible) that has a very bad bulge.

Maybe it's just over extrusion that only shows when you have to make a 100% fill?