supermerill/SuperSlicer #800: Speed is not adopted in thin places

SphaeroX, 2020-12-13:
### Version
2.2.0

### Operating system type + version
Windows 10

### Behavior
At a very thin spot (blue 19.3) I just noticed that the printing speed is very slow. So I wanted to search for this setting and have gradually put all the speed parameters up.

But I could not find any option that would have made it faster, is this a mistake on my part or is it the slicer itself?

[image]

[image]

tecnopolis-ve, 2020-12-13:
perhaps the printer cannot reach the top speed due to short movements.

SphaeroX, 2020-12-13:
Physically, maybe yes, but the slicer should still make the specification... 

I would already be satisfied with 30mm/s, the 100 were only for testing purposes

k2dp2k, 2021-01-02:
Same here on macOS. 
Tried with different printers, all show the same behavior.

supermerill, 2021-06-02:
There is now speed settings for thin things. Can you post your project file if it's not fixed?

closed 2021-06-04 by supermerill