supermerill/SuperSlicer #803: Temp offset of not working

bakaufman, 2020-12-15:
### Version
_Version: 2.3.55.2
Build: SuperSlicer_2.3.55.2
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Catalina Version 10.15.7 (Build 19H15)
Total RAM size [MB]: 17,180MB

### 3D printer brand / version + firmware version (if known)
Railcore Dual extruder hotend. Marlin bugfix 2.0.X from last thursday

### Behavior
two color print. Using Printer Settings>Offsets>Extruder temp offset> -10 for both extruders. Base temperature is 195 first layer, 190 other layers. I found the printer stuck at 180 on T0 and 195 on T1. No oscillation in the octoprint profile temp.

#### Project File (.3MF) where problem occurs
[PS-DUAL-cali_1-R.gcode.zip](https://github.com/supermerill/SuperSlicer/files/5693421/PS-DUAL-cali_1-R.gcode.zip)
https://discord.com/channels/771316156203270154/788215460147101707/788261045579153408

supermerill, 2021-03-30:
sorry to be that late. Can you post the project / .3mf file (zipped)?

supermerill, 2021-06-14:
you're using `M104 T0 S{first_layer_temperature[initial_extruder]}`
but you should have wrote `M104 T0 S{first_layer_temperature[0]+extruder_temperature_offset[0]}`
or `M104 T{initial_extruder} S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]}`
(not tested, I may have made a typo)

closed 2021-06-14 by supermerill