supermerill/SuperSlicer #806: Support interface prints in mid air

ProfEngr, 2020-12-16:
### Version
2.3.55.3 nightly 244

### Operating system type + version
Win7

### Behavior
Without the sheath enabled, the lower support layers don't close off the end in preparation for the interface layer. The sheath for this print would add nearly 3hrs & 13m of filament to fix the handful of places where I see this problem.

[image]
[image]

There must be some rounding errors in the algorithm that generates the support structure. The seemingly random appearance of gaps in the corners is 'interesting.' It seems to change based on the pattern spacing value. It's a shame Slic3r/PS didn't adopt a support algorithm that uses a percent infill instead of a grid spacing.

[image]

ProfEngr, 2020-12-16:
Oh, Is there an option to disable the bed interface feature if it's not necessary for adhesion?

Zulaner, 2020-12-16:
[image]

Zulaner, 2020-12-16:
[image]
[image]

ProfEngr, 2020-12-16:
@Zulaner i discovered this when trying to tweak the pattern spacing value. Above about 12mm I had support protruding above the part surfaces in the preview.

I also discovered, possibly, a new bug during the print. About 30% into the job I received a machine error as the nozzle was told to move a fraction of a mm outside the build volume. This was never flagged in the slacker preview.
I'll have to open a new issue on it after supper.