supermerill/SuperSlicer #808: Travel path extended beyond bed limits.

ProfEngr, 2020-12-17:
### Version
2.3.55.3 nightly 244

### Operating system type + version
Win7

### Behavior

While printing the part from issue 806 I was presented w/ the scenario shown in the pic. There was no warning from the slicer itself as when a part is larger than the build volume (235x235x250) set in the config. I had avoid crossing perimeters enabled. The part itself is 230x230x220. Pro Tip: remember to turn on travel moves in the preview when close to the build dimensions.

The sad thing is, the nozzle can easily move 5mm outside the configured volume, but it cannot print there.

[image]

[image]

ProfEngr, 2020-12-17:
I'm going to run this gcode again after removing the offending G1 entries manually. I verified via the new preview app that the changes wouldn't "break" anything. I tried reslicing the part w/o the avoid option, but ended up w/ cooling/warping issues that the original didn't exhibit.

supermerill, 2022-02-03:
I need the project file to reproduce the error, can you put it here?

OldCurmudgeon3DP, 2022-02-03:
I'll have to see if I still have the gcode file. I had a fit getting the file to print and ended up literally slicing it into pieces so it'd complete.

ashthespy, 2023-04-07:
I just ran into this using 2.5.59. 
[image]
Do let me know what you might need to help debugging.