supermerill/SuperSlicer #809: Bridge infill sometimes sticking out of model

tetele, 2020-12-17:
### Version
_2.2.53.4_

### Operating system type + version
_Win10 x64 19042.685_

### 3D printer brand / version + firmware version (if known)
_BLV MGN Cube running RRF 3.3.1_

### Behavior
* Sometimes, the bridge infill draws a line that's sticking way out of the printed object for no apparent reason. No other type of feature has displayed this behavior, only bridge infill. I have had it happen ~3-4 times in ~50 slicings
* _Steps needed to reproduce the problem_
  * unknown
* _Expected Results_: bridge infill stays inside the outer perimeter
* _Actual Results_: a single loop of bridge infill sometimes sticking out

[image]

#### Project File (.3MF) where problem occurs
[Spool roller 6001ZZ Side.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5708637/Spool.roller.6001ZZ.Side.3mf.zip)

neophyl, 2020-12-17:
That's not bridge infill , its the supporting dense layer I think.  Same issue as https://github.com/supermerill/SuperSlicer/issues/609 which I reported for version 2.2.54 which was fixed in a later release and as such closed.

tetele, 2020-12-17:
Right. I missed that because I was searching for "bridge infill", as the legend implies. But it does seem like a duplicate, indeed. Thanks for pointing that out.

closed 2020-12-21 by tetele