supermerill/SuperSlicer #810: Avoid crossing perimeters creates huge travel moves

ProfEngr, 2020-12-17:
### Version
2.3.55.3 nightly 245

### Operating system type + version
Win7

### Behavior
See attached image. With the option checked, the travel moves are jumping all over the bed unnecessarily. The majority of the blue lines dart across the open space in the middle only to return to the infill or part not far from where the move started (relative to the distance the avoidance algorithm took).

[image]

AeroNib, 2021-02-21:
I'm encountering this same bug when slicing objects with small islands. Between islands, the gcode will quickly jog the tool to just outside the overall perimeter at its nearest vertex. @ProfEngr's post shows it happens with supports, too, although I'm encountering it with no supports enabled. Maybe it's moving to some nearby point to avoid crossing the islands, but I haven't had a chance to look at the code to see how that algorithm works.

I'm running build 2.3.55.2.

This seems to be the same issue as slic3r/Slic3r#3721 which was opened in 2017, marked as potentially fixed but needed testing, and no updates after that.

**Here's with `avoid_crossing_perimeters = true`**
[image]

**and with `avoid_crossing_perimeters = false`**
[image]

supermerill, 2021-11-05:
If it's not fixed in the next version, please report.

closed 2021-11-19 by supermerill