supermerill/SuperSlicer #811: Maximum volumetric speed hardcoded?

Phil1988, 2020-12-17:
### Version
2.2.53

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Duet Wifi with RepRap Firmware 2

### Behavior
I am trying to print fast with an 0.8mm Vulcano nozzle.
I noticed, that it is not possible to print faster then 14mm³/s 

I have set this:
[image]

[image]

[image]

[image]

But the speed is somehow "limited" to 14mm³/s
[image]

Do I miss something here or is it internally set to maximum 14?

neophyl, 2020-12-17:
Zip up and post your project and we can take a look at it. That way we can see all your settings.

Phil1988, 2020-12-17:
how to do so?
save to project as 3mf and then zip it or zip a special folder or something else?

supermerill, 2020-12-17:
as you like, but you have to drag-drop a zip file here that contains your project file.

Phil1988, 2020-12-18:
[4github.zip](https://github.com/supermerill/SuperSlicer/files/5713983/4github.zip)

I hope this is what you wanted :)

neophyl, 2020-12-18:
When I load your saved project in you have the max volumetric rate on your filament profile set to 14. 

I have seen this before with a user on the prusa forum that was also using a multi extruder/filament setup.  Its one of the often overlooked problems with a multi filament setup. 
On the plater screen you have the drop downs that are used to assign SAVED filament profiles to each extruder.  When you go to the filaments tab and edit the filament profile, you are editing a sort of live copy.  With a single extruder setup the changes are applied 'live'.  As theres only one extruder the software knows it can apply the changed profile.  When using a single extruder setup you will see the  filament profile name in the drop down change to xxxxxxxx (modified) as its using the settings from the filament tab that have been changed.
However with more than one extruder it cant know which (if any) of the extruders to apply the (modified) filament profile to.  Notice on the plater that the filaments refer to the saved profiles.  You will have to SAVE the filament profile and assign it for any changes to be used.

Phil1988, 2020-12-18:
Thanks a lot neophyl!
This helped a lot and I didnt saw that before.

Its works and the issue will be closed now.

closed 2020-12-18 by Phil1988

Tinchus2009, 2020-12-21:
That is true! "problem" solved.

darksnakezero, 2023-05-30:
I ran into the same problem on 2.2.58 and it took me 5 days to figure out why my extruder was clipping.

Can we integrate a fat warning sign on top of the filament settings, as soon as multiple extruders are configured?