supermerill/SuperSlicer #812: rear positioned seam looks huge

delious87, 2020-12-17:
### Version
2.2.53

### Operating system type + version
Win7 64 bit

### 3D printer brand / version + firmware version (if known)
ZAV-MAX-PRO with Lerdge-K firmware 4.2.0

### Behavior
100*100*100 hollow cube sliced with rear seam prints with the giant seam because the X position not the same after layer change. 

There is g-code example from the middle of the print:

G1 **X150.060** Y149.760 E1.80626
; stop printing object 20mmcube.stl id:0 copy 0
G1 Z5.600000 F9000.000
; printing object 20mmcube.stl id:0 copy 0
G1 **X150.000** Y149.760
G1 F2400.000
G1 X100.240 Y149.760 E1.80844
G1 X100.240 Y50.240 E3.61689
G1 X199.760 Y50.240 E3.61689
G1 X199.760 Y149.760 E3.61689
G1 **X150.060** Y149.760 E1.80626
; stop printing object 20mmcube.stl id:0 copy 0
G1 Z5.800000 F9000.000
; printing object 20mmcube.stl id:0 copy 0
G1 **X150.000** Y149.760
G1 F2400.000
G1 X100.240 Y149.760 E1.80844
G1 X100.240 Y50.240 E3.61689

When i choose aligned seam the X position is the same after layer change.

SimonQuvang, 2024-01-03:
Project file not provided, so impossible to check the settings. However the X position difference is used to have a bit of overlap between the start and and the end of the line (0.60 mm as I understand). However it is very curious that the x should then be the same for aligned. 

Not sure if this issue should be kept since it's old and no project file was supplied @supermerill

supermerill, 2024-01-16:
There is always a gap for the seam, the seam position doesn't impact that. Even with "aligned seam" you should see a gap (maybe in y? i don't knwo, i don't have your project)
But you can change it via the seam_gap setting.