supermerill/SuperSlicer #814: [Feature request] Dark mode on Windows 10

vidpugelj, 2020-12-18:
The title says it all. This would be a really welcome addition as it seems that Mac users already have it.

Thanks for all the work you do on this slicer!

novarlynx, 2021-05-15:
Would love to have this too.

supermerill, 2021-05-20:
duplicaye of https://github.com/supermerill/SuperSlicer/issues/413

closed 2021-05-20 by supermerill

DRA6N, 2022-02-05:
+100000 to this. Prusaslicer is built on the old Win32 API and the devs over at Prusaslicer use that as an excuse to not build a workaround. Isn't the point of Superslicer to be better? So be better, give us darkmode! :)

supermerill, 2022-02-06:
Prusa has created a workaround in 2.4
It's a huge dev effort, as it means recreating many widget so they can be colored the right way.
No tab, no status bar, no dialogs, etc.