supermerill/SuperSlicer #815: Maximum layer height not enforced

ardyesp, 2020-12-18:
### Version
2.3.55

### Operating system type + version
Windows 64

### 3D printer brand / version + firmware version (if known)
Custom CoreXY

### Behavior
The parameter ```printer: max_layer_height``` is not enforced in the ```print: layer_height``` configuration. When someone enters a layer height bigger then ```max_layer_height```, the slicer should give a warning message, similar to one given when the ```layer_height > nozzle_diameter```.

[image]

neophyl, 2020-12-18:
Have you actually checked what the produced layer height is though when sliced ?

ardyesp, 2020-12-18:
> 
> 
> Have you actually checked what the produced layer height is though when sliced ?

Yes

[image]