supermerill/SuperSlicer #816: First layer temperature bug?

Tinchus2009, 2020-12-18:
### Version
2.3.55
### Operating system type + version
Win10 64

### 3D printer brand / version + firmware version (if known)
Custom corexy reprap foirmware
### Behavior
I have created a printing profile and a material and I want to use the temperature calibration feature. So, in my material profile, default first layer temperature is 270

I set the temp tower values to have 2 steps above that and 4 steps below that temp, in increments/decrements of 10 degrees.

The resulting code for some reason has a first layer temperature os 200 degrees. I use this setting for setting my first layer temperature on my start gcode settings:

G10 S{first_layer_temperature[0]}

This command is creating a G10 S200

And I don't know where the 200 is coming from.

Attached are the code and the config file and the generated code file

[temp.zip](https://github.com/supermerill/SuperSlicer/files/5718238/temp.zip)

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

Tinchus2009, 2020-12-18:
I forgot the gcode
[temp_calibration.zip](https://github.com/supermerill/SuperSlicer/files/5718271/temp_calibration.zip)

neophyl, 2020-12-19:
I loaded that configuration in and the filament first layer temperature is set to 200, the filament temperature above the first layer is ALSO set to 200.  If you read the raw config in a text editor you can see it is set to 200 for both even without loading it in.

I notice you are using a multi extruder setup.  Please have a read of https://github.com/supermerill/SuperSlicer/issues/811 as it is possible you are making the same mistake but with your temperatures.

Tinchus2009, 2020-12-21:
Yu are right! "problem" solved!

closed 2020-12-21 by Tinchus2009