supermerill/SuperSlicer #817: Ironing how to enable?

aceycv8, 2020-12-19:
### Version
2.2.53

Hi Guys,

This is probably a user issue more than anything, but how do i enable ironing?
the only settings i see are in - inflill- "advanded infil" iron tuning but it's all greyed out.
[image]

i can't right click the model and add in a new modification to add in ironing etc so i assume it's something i've enabled along the way but i dont know what?
[image]

Wil-Storch, 2020-12-19:
"External pattern Top" must be set to "Ironing" in the version you are using.

Later versions this functions slightly differently
[image]

aceycv8, 2020-12-19:
Thank you Wil-Storch!

Phil1988, 2024-05-28:
@aceycv8 or @supermerill 
This can be closed.

closed 2024-05-28 by supermerill