supermerill/SuperSlicer #818: Disable M600 at start of print

khthe8th, 2020-12-19:
### Version
2.3.55.2

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro with Klipper

### Behavior
Doing single extruder multimaterial setup, I have filament swap halfway through print but no matter what settings I seem to try SuperSlicer adds an addiitional tool change Gcode before the start of the print. I already have my start filament loaded, so I have been manually editing the gcode to remove this tool change at the start.
Did I miss a setting somewhere? 

It properly has a toolchange gcode halfway through my print (minus the error with not being able to name my extruders without SS crashing due to "bad cast" but I saw that is being fixed in the next update) 

#### Project File (.3MF) where problem occurs
`;TYPE:Custom
; custom gcode: start_gcode
START_PRINT T_BED1=65 T_BED2=60 T_EXTRUDER=210
; custom gcode end: start_gcode
M109 S210 ; set temperature and wait for it to be reached
M190 S65 ; set bed temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; custom gcode: toolchange_gcode
M600
; custom gcode end: toolchange_gcode
ACTIVATE_EXTRUDER EXTRUDER=extruder
G92 E0
; custom gcode: start_filament_gcode
; Filament gcode
; custom gcode end: start_filament_gcode
;LAYER_CHANGE
;Z:0.28
;HEIGHT:0.28
G1 Z0.280000 F12000.000
G1 E-1.80000 F2100.00000
G1 X81.246 Y80.572 F12000.000
G1 E0.00000 F1200.00000
;TYPE:Skirt`

neophyl, 2020-12-19:
No you haven’t missed a setting. It’s the same in Prusa Slicer. How about not loading your filament in before hand. ?  As the printer/slicer can not know you have the correct filament loaded it’s not really doing anything it shouldn’t. 

If the behaviour is changed then that could potentially mess up someone’s existing workflow. Also as the gcode could be run multiple times it’s not guaranteed that it would always be correct before hand.

khthe8th, 2020-12-19:
> No you haven’t missed a setting. It’s the same in Prusa Slicer. How about not loading your filament in before hand. ? As the printer/slicer can not know you have the correct filament loaded it’s not [...]

Interesting points, but then by that logic how come when I do a normal single color print it doesn't do an M600 at the beginning to make sure I have the color I want loaded before it starts the print?

cycletronic, 2021-01-12:
Another problem with leaving it as-is, and "not loading filament beforehand" is that the Start G-code section is inserted before the M600 tool change command.  Start G-code in my case includes the intro/prime line.  

It seems possible to make a workaround, but I would also like to see an option to skip the startup M600 command.

Loreagle, 2021-01-18:
I agree with you, multimaterial with single extruder is very beautiful because you can use materials with different temperatures instead of a simple filament change but that M600 at the start of the print is useless on 90% of cases. A box as you suggested would be perfect to cover 100% of cases.

supermerill, 2021-06-02:
if you don't have a filament change in the first layer, you can use `{if layer_num > 0}M600{endif}`

Loreagle, 2021-06-04:
> I found a solution: you can use `{if layer_num > -1}M600{endif}`

Perfect thank you!!

I have another question related: why when using normal color change I can see statistics for each filament and instead when using Single Extruder Multi Material I can only see one total stat???

supermerill, 2021-12-17:
> I have another question related: why when using normal color change I can see statistics for each filament and instead when using Single Extruder Multi Material I can only see one total stat???

I think there ere a bit borked by the 2.3 merge. Should be fixed in the nightly.

ibrushd1, 2022-01-26:
supermerill: i want to disable this filament swap at the beginning as well but where do i place the {if layer_num > -1}M600{endif} in the firmware? this tool change is really frustrating because I have my printer all ready to print and then it goes into the filament swap and retracts the filament then the head cools off slightly and when it feeds the filament back in again it wont purge and just gets clogged and creates problems and also sometimes i am repeating the same print and dont need to change filament yet it still prompts a filament change! just frustrating the hell out of me! please advise where to insert that fix and thank you in advance!

supermerill, 2022-01-26:
> supermerill: i want to disable this filament swap at the beginning as well but where do i place the {if layer_num > -1}M600{endif} in the firmware?

in the `tool change gcode`, you can search it with the search tool (magnifying glass)

positron96, 2022-03-13:
How does the app decide when to emit M600?
I have 2 files sliced with exact same SuperSlic3r 2.3.57 and one has M600 at start, the other doesn't. The settings differ, but since I have 1 extruder on Ender 3 and have never done multi-material prints, I have never fiddled with settings related to filament/tool change.

positron96, 2022-03-13:
Ah, never mind, it's color change command that's inserted whenever one misclicks on layer slider in preview

Rhodani, 2023-12-14:
> I found a solution: you can use `{if layer_num > -1}M600{endif}`

where would I add this line of code?

neophyl, 2023-12-14:
https://github.com/supermerill/SuperSlicer/issues/818#issuecomment-1022388237