supermerill/SuperSlicer #822: [request] wall thickness recommendations

Phil1988, 2020-12-21:
### Version
2.2.53

### Operating system type + version
Win10 x64

### Behavior
This is not a bug or an issue but more a "feature request" :)

I found it a little bit confusing how SuperSlicer recommend the wall thickness.

I know what it means but I think it would be benefitial to show the recommended thickness of the walls similar to PrusaSlicer.

The reason is that, if you are designing an object its sometimes also important to have "uneven" wall counts.
The way SuperSlicer shows it is only for even wall counts (perimeters).
But it could save a lot of time if you design an object with 3 walls that perfectly lines up.
In this case, 2 walls could be too weak and 4 would be not necessary.

To demonstrate it, here is a picture of PrusaSlicer left and SuperSlicer right:
[image]

Is it possible to either change this similarly or do add something like PrusaSlicer has?

That would make it more usable in cases where you're designing something speed optimized :)

As me and others are mostly writing for issues or requests:
This slicer is really great and I would like to thank everyone working on it :)

supermerill, 2020-12-27:
Maybe it's now possible to have a kinda-3 perimeters if the gap fill is changed into a perimeter, but I won't put a bet on it.

When you have a loop, you have to count it both ways. You see 2 perimeters but it is one loop. You can't have 3 periemters.

Phil1988, 2020-12-29:
Technically, you are absolutely correct if you focus on the definition of the word "perimeter".

The point is, that if you're constructing an object for the purpose of beeing 3D printed and doing it seriously, you will have to take several aspects into account.

One of these aspects is the quality you want to have as the result - and therefore the perfect line width for the height you want to print.
It will not only improve the quality if you set the width/height ratio correctly but it will also reduce the printing time dramatically if you're designing the model in the right way. I am talking about at least 20% time saved for an "OK" model to an optimized model and 40%+ for an "bad" designed model vs an optimized. And this saved time is while the quality is the same or most likely even better (because perfect layer width also results in a more homogenious surface).

So basically I am asking about something that shows the user what the perfect horizontal wall thicknesses would be for the set height and line widths.
I know there is something called "gap fill" but in a perfect print this would have the same width as the perimeters to perfectly bind. 

An example would be the face shields against corona. With perfect wall sizes you can lower the print time for a standard printer from about 3hours to 50min with great results. But for this you need to know what wall size your model should have.
This is a real example that I experienced and is something about 72% time reduction - still while having even better results.

Lets assume you are designing a complex model that will need different wall sizes because of looks and stability.
It would be very handy if the slicer could tell you "a single line would be 0.45mm", "2 lines 0.82mm", "3 lines 1,21mm"... etc.

Me as a user, I dont want to calculate 30min for the given values, what wall thickness would match the perfect line width (regarding the line width itself plus the overlap of each line and the overlap to the infill etc...). 
And if the digits/wall thicknes are not suitable for the model, I have to change the line widths and do the calculation again...

If the slicer can just tell me "hey, for your given values, the perfect wall size would be 
A mm for 1 line, 
B mm for 2 lines, 
C mm for 3 lines,
etc.

This would be great and a huge benefit for many users. 

I hope you understood the point and I could convince you, that this would be a great function ;)

Tahx, 2022-02-06:
@supermerill I'm pushing for this, right now it's simply confusing, the tooltip doesn't help. Gap fills bring their own kind of issues and uneven walls should absolutely be there.

supermerill, 2022-02-09:
> If the slicer can just tell me "hey, for your given values, the perfect wall size would be
> A mm for 1 line,
> B mm for 2 lines,
> C mm for 3 lines,
> etc.
>
> This would be great and a huge benefit for many users.

but currently, you have 
```
Recommended object min (thick) wall thickness for layer height 0.28
and 1 perimeter: 1.74mm
and 2 perimeters: 3.54mm
```
that is exactly that.
I can reword the help with your sentence.
```
Recommended object min (thick) wall thickness for layer height 0.28
and 1 perimeter: 1.74mm
and 2 perimeters: 3.54mm
```
[image]