supermerill/SuperSlicer #824: Bridges being detected as overhang perimeters

Tinchus2009, 2020-12-21:
### Version
2.3.55
### Operating system type + version
win10 

### 3D printer brand / version + firmware version (if known)
custom reprapfirmware

### Behavior
Look at the picture. The perimeter that, I think< should be detected as bridges, are being treated as overhangs. The inner perimeters are treated as bridges as expected.
The problem with this is that the values you expect to be used on briges are not being used on those external perimters.

[image]

0x4d4e, 2021-03-17:
Maybe related - just the other way around: what should be an overhang perimeter is detected as bridge-infill

v2.3.55
[image]

v2.2.53
[image]

supermerill, 2021-03-18:
@0x4d4e : i need the project file, please.

@Tinchus2009: currently, all unsupported perimeters are set as "overhangs". I can change that so that all continuous strait overhangs are now tagged as "bridge", but that will change some overhangs like the ones in the top of the "//" of the voron cube to be bridges, are they are strait. 

I don't see how I can put them apart. maybe with a third threshold....

0x4d4e, 2021-03-19:
@supermerill zipped project file attached. Thanks!

[Voron_Design_Cube_v7_bridge_infill_overhangs.zip](https://github.com/supermerill/SuperSlicer/files/6172588/Voron_Design_Cube_v7_bridge_infill_overhangs.zip)

closed 2021-11-07 by supermerill