supermerill/SuperSlicer #825: filament flow calibration breaks bl touch probe

raymwe, 2020-12-22:
Version 1.2.5.3
w10
ender 3 pro

Printed the flow calibration, the height of the blocks are above retracted posn. of bl touch probe. This needs a serious warning message, or separate the blocks further apart, or print singly. snapped off the probe tip.

bangor-flyer, 2020-12-22:
No issue with my BL touch done 3 levels yesterday.
This would mean your probe is below your nozzle poss?

raymwe, 2020-12-22:
No, It is retracted, just above nozzle, but this test prints a block at a time, so the block height is greater than the difference between retracted probe tip and nozzle tip. Also, it will collide with cooling fan ducting. If you want to print an item at a time, it needs to have enough distance between the items. This does not. Afaik, there is no specified distance for probes to be above the nozzle. Just because it worked OK for you, it does not mean that the test is safe for others. fwiw, the earlier bl touches had a shorter retraction distance.

bangor-flyer, 2020-12-22:
[image]

raymwe, 2020-12-22:
The information on setting up the printer for flow calibration is not adequate. If others think it is, so be it. I've reported a problem. It would be trivial to correct it.

closed 2020-12-22 by raymwe