supermerill/SuperSlicer #827: flow calibration / 1/4 wall thickness is off.

BelgianGuyWithALaser, 2020-12-22:
### Version
2.2.53

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version
CR20 (modded) + klipper

### Behavior
Printing a hollow cube for flow calibration results in a non even print. 3 walls are nearly the same thickness. while the 4th is thicker. Printed the same STL with cura and the problem was gone there.
(wall thickness with 100% flow also was different from cura to superslicer. so taking those old value's where a no go...
Printspeeds / extrusion multipliers where set exactly the same. yet resulted in different wal thicknesses. )
[image]
Measured with a .01mm micrometer.

#### Project File (.3MF) where problem occurs
[flowcube.zip](https://github.com/supermerill/SuperSlicer/files/5731740/flowcube.zip)

mucomuco, 2020-12-23:
I have same problem.

ardyesp, 2020-12-23:
Slic3r uses a different [model for extrusion](https://manual.slic3r.org/advanced/flow-math), vs Cura's rectangular approximation, so printed wall thickness will be different.

Regarding the one wall being different from other three, have you compared the 4 walls gcode to see that SS is indeed putting out more plastic on one side. I did this sample comparison and see that all four sides are same (smaller extrusion get proportionately less):
```
                                    Extrusion   Filament
                                    Length      Used
;LAYER_CHANGE
;Z:1.3
;HEIGHT:0.2
G1 Z1.300000 F10800.000
G1 X113.207 Y136.793
;TYPE:Perimeter
G1 F1648
G1 X113.207 Y113.207 E64.21923      23.59       0.94661
G1 X136.793 Y113.207 E65.16584      23.59       0.94661
G1 X136.793 Y136.793 E66.11245      23.59       0.94661
G1 X113.267 Y136.793 E67.05666      23.53       0.94421
G1 X112.750 Y137.250 F10800.000
;TYPE:External perimeter
G1 F1648
G1 X112.750 Y112.750 E68.03996      24.50       0.98330
G1 X137.250 Y112.750 E69.02326      24.50       0.98330
G1 X137.250 Y137.250 E70.00655      24.50       0.98329
G1 X112.810 Y137.250 E70.98745      24.44       0.98090
G1 X112.950 Y136.904 F10800.000
```
Since gcode is not instructing the printer to deposit extra filament on one side, this leaves us with printer mechanics as the possible cause. Usually this happens to the first printed wall after the layer change. Pressure advance and slicer's wipe settings might also influence it.

Also, since slic3r model overlaps the walls, measuring the wall thickness for flow calibration is not the correct. Use [Triffid Hunters](https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide#E_steps) method or built in flow calibration prints. Measuring wall thickness of printed cube can never yield good result since the layers don't stack up perfectly - so you will end up measuring the Z wobble and extrusion inconsistencies. See this [picture](https://jgauroraforum.com/discussion/comment/4568/#Comment_4568) of layers under a microscope.

BelgianGuyWithALaser, 2020-12-23:
> Since gcode is not instructing the printer to deposit extra filament on one side, this leaves us with printer mechanics as the possible cause. Usually this happens to the first printed wall after [...]

I can rule out my printers mechanics already since it's perfectly the same on the cube printed with Cura. if it would be the printers mechanical setup it would pop up in there aswell. already even tried S3D and prusaslicer and those 2 also give me 4 walls that are close to the same thickness while only in SS it doesn't.
the Triffid page doesn't mention anything about flow calibration, and the build in flow calibration is "meh" since it's also depending on if your bed is level. your Z-offset is correct etc etc.
I've set it so that slicer shouldn't overlap the walls. so the thickness I'm measuring should be the correct one that way.

I'll check the G-code later if something is in there.