supermerill/SuperSlicer #829: Feature Request: wait for the first_layer_bed_temperature during sequential printing

FillPavS, 2020-12-24:
When printing several objects sequentially, if first_layer_bed_temperature and bed_temperature are different, after finishing printing the first model, need to wait until the bed warms up to first_layer_bed_temperature. Instead of going to print the second model right away with bed_temperature.

kursatu, 2020-12-25:
You can do that via custom gcode.
This is what I've been using to achieve that.

Insert the following code into the 
Printer settings/Custom G-code/Between objects g-code (for sequential printing)

; Wait for temperatures
M140 S{first_layer_bed_temperature[initial_tool]}  
M104 S{first_layer_temperature[initial_tool]}
M106 S0
G1 Z0.1 E-5 
G4 S15
G1 E5

FillPavS, 2020-12-25:
Yes. Thank you. Didn't look there

closed 2020-12-25 by FillPavS