supermerill/SuperSlicer #831: [Feature Request] Infill angles

wildjokerss, 2020-12-25:
### Behavior
* _Describe the problem_ Currently SS only allows to set a certain degree of rotation for infill, problem is that this rotation affect all the infills, either solid or not. This is far from ideal. 
* _Expected Results_ I would like a approach similar to Ideamaker, with a box where I can add the degree of infill I want and slicer does it by that order, per layer. Also, splitting infill from solid infill, I might want my solid fill at 0 or 90º angle, or even 0º then 45º then 90º and leaving my standard infill untouched.
* _Actual Results_ When using the rotation of the infill this is affecting every infill, so it kinda "breaks" some of them by making them very weak.
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]
[image]

andybrice, 2021-06-22:
Yes, this is something I often find myself wanting too. I might want my top and bottom faces angled a certain way for aesthetics, and my infill angled another way for strength.

Perhaps it could be achieved with something like this? (Although it would add clutter.) [image]

marcel-isler, 2022-06-24:
I would love to see this feature added as well... it's one of the main reasons I have not switched from S3D yet...

I have my infill angles set like this:
[image]

This gives me great coverage, especially over low % infill.  It's also much less likely to have a checkerboard pattern.

closed 2024-02-11 by supermerill