supermerill/SuperSlicer #834: Starting Gcode Ignoring M106 commands

3Mistakes, 2020-12-27:
### Version
2.3.55.2

### Operating system type + version
Win10

### Behavior
Super Slicing is ignoring all M106 commands in Starting Gcode.
[image]
From the .gcode file:
;TYPE:Custom
; custom gcode: start_gcode
M104 S150 T0; Bring nozzle to 150C
M109 S150 T0;
; set and wait on bed temperature
M140 S50
M190 S50
; set and wait on hot end temperature (Finish raising the hot-end temp to set value)

jhstatewide, 2021-01-21:
@3Mistakes 

Workaround:
Set Cooling fan under printer settings.
Kickstart duration = 0

I had the same problem and it was tricky to diagnose.

closed 2021-11-07 by supermerill

YairHH, 2022-08-06:
Hi

I just encountered this same behaviour with version; 2.3.57.12

The only reason I noticed is that when I print from the SD card on my Ender 6, the firmware always turns on the cooling fan on print job start; I saw it did not turn off at all, although my start code uses command M107 and M106. Printing from Octoprint is not an issue as the machine does not automatically turn on the fan on print start.

 After inspecting Gcode I also noticed the custom lines not appearing at all. 

I found this thread when looking for answers, and after setting kickstart and speed up to 0 as suggested, my custom Gcode lines are now present and when printing from SD card fan indeed turns off. 

I like the kick start and speed-up features; I don't want to sacrifice them just to have first-layer fan control. 

Thanks.

supermerill, 2022-08-06:
fixed in 2.4
I don't think I'll move it to the 2.3 branch, to much hassle.