supermerill/SuperSlicer #835: Slicer leaves gaps in solid layers and whole model

EisbaerLars, 2020-12-27:
### Version
SuperSlicer 2.2.53

### Operating system type + version
Win10 20H2

### 3D printer brand / version + firmware version (if known)
Ender 3 with Duet2 Wifi 3.1.1

### Behavior
When slicing a STL the slicer view shows first the expected first layer fully closed/solid, but a second later the slicer shows a first layer with gaps - this rules for all layers since the gaps are reproduced in every layer through the whole model

Have attached a project file with the behavior as well as a picture and videos showing whats happening with above described behavior

* _Expected Results_
Normal slicing with closed/filled surface

* _Actual Results_
This is the first layer after slicing - with a lot of gaps and this gaps run through the whole model
[image]

This video shows the behavior for the first attempt to slice the model
https://user-images.githubusercontent.com/59581026/103181532-353ca200-48a2-11eb-83ae-32b54e44d3a4.mp4

Next video shows the same slicing but with first layer view - for a second the correct slicing appears but gets screwed up with what looks like a second pass
https://user-images.githubusercontent.com/59581026/103181551-78971080-48a2-11eb-8dbc-29134c01330a.mp4

#### Project File (.3MF) where problem occurs
[Bug Report.zip](https://github.com/supermerill/SuperSlicer/files/5745385/Bug.Report.zip)

EisbaerLars, 2020-12-29:
SOLVED - found the issue by myself.
In Printer settings/General the Minimum extrusion length was much too high

closed 2020-12-29 by EisbaerLars

jrpmedia, 2023-10-20:
Where is this 'Minimum extrusion length' in v2.3.57 please?