supermerill/SuperSlicer #840: Gcode Preview doesn't display extrusion width above 1mm

bung69, 2020-12-29:
[image]

### Version
2.3.55.2

### Operating system type + version
Win10

Gcode Preview doesn't display extrusion width above 1mm.   attached screenshot is with a 1.4mm nozzle with auto extrusion width.  confirmed with manual extrusion widths set > 1,4mm

liftbag, 2021-01-04:
I can confirm this behavior. It almost feels like a limit set in the slicing engine. In fact, if you set an extrusion width greater than 1mm, a dialog box appears to confirm the choice. But even confirming the choice, refusing to use the percentage, the extrusion width cannot exceed 1mm.

[image]

[image]

supermerill, 2021-06-02:
it's just a min(1, ....) somewhere as a failsafe that was a bit too broad.
I also changed the check message, for some fields. If it's still a bit too trigger happy, please let me know to change it.

bung69, 2021-06-03:
Looks good now.  thanks

closed 2021-06-03 by bung69