supermerill/SuperSlicer #841: Crashes on slicing with gyroid at low infill % when using infill anchor

bung69, 2020-12-30:
[350pc benchy.zip](https://github.com/supermerill/SuperSlicer/files/5753522/350pc.benchy.zip)

### Version
2.3.55.2

### Operating system type + version
win10 pro 1803

### Behavior
slicing a large 350% benchy with gyroid infill crashes at low infill percentages and infill anchor set to 1000.    have also experienced this before with other models.  

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

neophyl, 2020-12-30:
That version of Susie is built against an older alpha version of Prusa Slicer.  That version of PS has an issue with the new infill anchor method they came up with.  You need to not use the Prusa anchor method in that version (0 simple connect) which makes it use the non PS method it used in previous versions.   
It was fixed in PS since then but Supermerill mentioned that he wasn't going to merge against PS again until they had sorted out the issues in PS and done a  proper release.

bung69, 2021-01-03:
ah ok, thanks for clarifying.

closed 2021-01-03 by bung69