supermerill/SuperSlicer #843: Random crash if using any type of supports on certain .STL

BennyGP, 2020-12-31:
**### Version**
SuperSlicer_2.3.55

**### Operating system type + version**
Win 10 Pro

**### 3D printer brand / version + firmware version (if known)**
Self builted prusa style, Marlin FW 2.XX

**### Behavior**
During the slice with some kind of STL (i don't know what does cause this) the program crash during the "Support generator" time.
Maybe is just a problem caused from the support gen for this model.
Normally i have the well done slicing and do issues.
But only some stl gives me this strange crash error.
// No screenshot available.

#### Project File (.3MF) where problem occurs
[Bug problem.zip](https://github.com/supermerill/SuperSlicer/files/5757565/Bug.problem.zip)

neophyl, 2020-12-31:
v2.3.55 is based on the alpha versions of Prusa Slicer, which have LOTS of bugs.  These of course come over to Susie.

I would be wary of using it given that fact.  Supermerill mentioned that they were not going to do another merge until there is a release from Prusa as it was too much work given the changing state of the PS versions.

vcapra1, 2021-01-08:
I'm having this issue too.  When I try to slice a simple 20mm cube with supports (obviously no supports are needed, but nevertheless they are turned on), it crashes immediately after slicing.  With supports off, it does not.  The exact same model in PrusaSlicer does not have this issue.

supermerill, 2021-06-02:
prusa is rewriting the support code, will look at that after if it's not fixed.