supermerill/SuperSlicer #846: Bed wants to heat up to nozzle temp at start!

hagan66, 2021-01-03:
### Version
2.3.55.2

### Operating system type + version
MacOS Big Sur

### 3D printer brand / version + firmware version (if known)
Prusa MK2.5S MMU2s

### Behavior
I think its a typo in the start gcode. Will try fixing it myself.

supermerill, 2021-01-10:
If it's not present in the start gcode, i add it myself. to prevent problems and people making issues because it doesn't heat.
If you don't want it, put a M109 in your start gcode. Even a M109 S0 works to remove that extra check.

supermerill, 2021-01-10:
please post your project file.

hagan66, 2021-01-10:
[xyzCalibration_cube.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5792784/xyzCalibration_cube.3mf.zip)

I fixed the problem by changing the start gcode from this:

M140 S[first_layer_bed_temperature] ; set bed temp
M190 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; wait for bed temp

to this:

M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp

hagan66, 2021-05-27:
This is still a problem in the latest release.

supermerill, 2021-05-27:
what you have in your star gcode:
```
M104 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; wait for bed temp
M109 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; wait for extruder temp
```
what you want:
```
M104 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; set extruder temp
M190 S{first_layer_bed_temperature] ; wait for bed temp
M109 S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]} ; wait for extruder temp
```

hagan66, 2021-06-01:
Hi again! Thanks for your suggested g-code. I have it working. :-) My concern was that the default start g-code is wrong (maybe just for my printer model?) and could cause overheat problems for other users. Every time I reinstall Superslicer this comes back, that the bed gets the nozzle temp in the start code.

supermerill, 2021-06-01:
If the prusa default settings bad?
if so, it's a serious bug.

closed 2021-06-13 by supermerill