supermerill/SuperSlicer #847: User defined seam position

ufoDziner, 2021-01-04:
### Version
Current beta

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Bear and Voron.  Both using klipper.

### Behavior
Would like to be able to specify the seam position in a more specific fashion than "Align" or "Rear" such that the seam may be aligned with a specific corner.  Maybe a dial/clock face similar to in Kisslicer would make it easy.  Thanks

Stephan3, 2021-01-04:
theres a draw mode where you can place the seam. Afaik its the most customizeable seaming among all slicers on prursa and superslicer.

ufoDziner, 2021-01-04:
Thanks, I see it now.  It's on the left menu of the "Plater" view called "Seam painting".  You're right, it looks amazing!

closed 2021-01-07 by ufoDziner