supermerill/SuperSlicer #850: Retract on layer change in Filament override enabled even with Vase mode

Kveri, 2021-01-04:
### Version
2.3.55

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro, newest Marlin

### Behavior
Retract on layer change settings is possible in two places: Printer profile and then as an override in Filament profile. Let's ignore Filament override for a second and assume Retract on layer change is defaulted (no override) in Filament profile. In such case, if printer profile has retract on layer change = enabled AND Print profile has vase mode = enabled then print is just fine - there is no retraction whatsoever, in the background it seems that vase mode disables retract on layer change, which is the intended behavior. So Vase mode acts like a sort of override for printer profile setting, although this is not specified in the popup info when enabling vase mode.

Now, let's add filament override into the mix. My default printer profile has retract on layer change = enabled. One of my filaments also has **force** retract on layer change = enabled (technically overriding enabled with enabled but in a different place). Other filaments don't override this setting (e.g. it's defaulted to enabled from printer profile). With Vase mode this creates a problem. If retract on layer change is force enabled in filament overrides then it is also enabled if Vase mode = enabled. There are no "real" layers in Vase mode, however it seems that there are sort of simulated layers and on each of these simulated layers the retraction happens exactly once.

The result is a seam-like spiral visible on the print.
[image]

Maybe I'm wrong, but from my point of view Vase mode should override even Filament overrides "retract on layer change". I think so because vase mode cannot ever benefit from retract on layer change = enabled, because there are no real layers in Vase mode.

Summary:
Vase mode correctly disables retract on layer change if retract on layer change is enabled in printer profile. However Vase mode doesn't seem to disable retract on layer change if retract on layer change is enabled in Filament overrides. 

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

liftbag, 2021-01-04:
I have tried to check it on a simple cylinder, and I cannot reproduce the problem. I don't see any artifacts in the g-code preview by forcing retraction at layer change in filament override.
Although it seems unlikely that it depends on geometry, it would be useful to check it on your stl.

Kveri, 2021-01-06:
> I have tried to check it on a simple cylinder, and I cannot reproduce the problem. I don't see any artifacts in the g-code preview by forcing retraction at layer change in filament override.
> Although it seems unlikely that it depends on geometry, it would be useful to check it on your stl.

I just tested it using a simple cylinder as well and I can replicate it. 
[image]

Here is my printer extruder profile - basically all default, just retraction length set to 1mm.
[image]

Print profile is standard builtin 0.20mm NORMAL @CREALITY just with Vase mode enabled (and of course all things what Vase mode enables automatically).

Filament overrides are here:
[image]

Could you compare it to your settings? I'll now try to start from absolute default profiles and only modify minimum amount of things and see.

Kveri, 2021-01-06:
OK. Now I simplified it heavily:

1. Default printer profile (System preset): Creality Ender-3
2. Default 0.20mm NORMAL CREALITY print profile with ONE modification: Spiral Vase = enabled (+automatic changes)
3. Default System preset Creality PLA filament profile, with ONE modification: Filament Overrides -> Retract on layer change = enabled.

With these settings I get this:
[image]

It's even worse than with my profiles but that may be due to other settings.

While if I just set the the filament override "retract on layer change" to disabled (override enabled, but overriding to disabled) I get this  (correct behavior):
[image]

Same result if I disable the override altogether (correct behavior).

JohanAR, 2021-01-19:
Sounds like this issue in PrusaSlicer: https://github.com/prusa3d/PrusaSlicer/issues/5004

Log has link to commit with fix for it.

liftbag, 2021-01-20:
Ok, I can confirm the problem. But, according to my tests, the error only occurs if both settings _"Retract on layer change"_ and _"Whipe while retracting"_ are enabled in the _Filament override_ section.

JohanAR, 2021-01-20:
@liftbag I can reproduce it with only "retract on layer change". At first I didn't realize I had to check the boxes both to left and to the right of it though. Could it be possible that you've missed the one on the right?

liftbag, 2021-01-20:
Only with these filament overrides I can reproduce the issue.
[image]

If I uncheck _Whipe while retracting_ ...
[image]

... or _Retract on layer change_, the artifact disappears
[image]

This is completely independent of the settings in _Printer settings._

JohanAR, 2021-01-20:
What is your printer settings -> extruder -> "wipe while retracting" set to?  I'm using default Ender-3 profile where it already is ON.

If I override "wipe while retracting" to OFF in filament settings (while still overriding "retract on layer change" to ON) I still get retractions in vase mode, but they all end up aligned in a corner so they are not visible unless I enable show retractions in the preview.

liftbag, 2021-01-20:
> But also since this seems to be fixed in PrusaSlicer, I suspect that it will be fixed here too once SuperSlicer pulls in those commits, so maybe we don't need to investigate further :)

I hope I can use the next release, because all the latest nightly builds don't open in macOS 11.

supermerill, 2021-06-02:
> But also since this seems to be fixed in PrusaSlicer, I suspect that it will be fixed here too once SuperSlicer pulls in those commits, so maybe we don't need to investigate further :)

is it fixed?

JohanAR, 2021-06-03:
> > But also since this seems to be fixed in PrusaSlicer, I suspect that it will be fixed here too once SuperSlicer pulls in those commits, so maybe we don't need to investigate further :)
> 
> is it fixed?

I can no longer reproduce on my latest build from master, but I discovered something new: Changing Filament Overrides -> Retract on layer change does not flag the model as needing re-slicing. It doesn't matter if I override to on, override to off, or enable/disable the override, the button in gcode preview still says "export g-code".. If I change any other override the button changes to "slice now".. Instead I used the re-slice now menu option to verify this