supermerill/SuperSlicer #852: Increase decimals for x-y axis

chriskoups, 2021-01-05:
This is feature request.

Most modern printers have 50 steps/mm or even more for the x-y axis which allow them a resolution of more than 0.020mm.
The current slicer produce a code with a resolution of 0.001mm which in theory is enough but can introduce rounding problems.
I will suggest to increase that resolution to 5 decimals like the z-axis and e-axis.

As an example, I have a CR-6, the steps/mm are 80, 80, 400 and 93 for x, y, z and e respectively.
That make the resolution 0.0125, 0.0125, 0.0025 and 0.01075269.
Currently, the gcode resolutions are 0.001, 0.001, 0.000001, 0.00001.
That produce a rounding of 5x10<sup>-4</sup>, 5x10<sup>-4</sup>, None and 2.69x10<sup>-6</sup>.

I can imagine that something that is 1μm right/left up/down is pretty small but may fix problems with alignments.

qwewer0, 2021-01-05:
What would be the down side to the finer resolution?

4% rounding isn't too high, but if other than the increased slicing time there isn't too much of a down side to the increased slicer precision, then would it be possible to add a new setting to adjust those values on the fly? So that if a user wants it, can live with the negative sides, while others wouldn't suffer a bit.

chriskoups, 2021-01-05:
I am not even sure if the slicing time is increased as I can imagine that the calculation behind the scene would be happening with more resolution. I think is the way that the value is exported which will introduce rounding.

The only downside that I can see is increase in gcode file by around 15% as all the commands that involve x and y must increase by 2 decimals so in a typical line instead of having 25-30 characters they will be 29-34.

And yes, they can have a setting as they have for z (z full step).

supermerill, 2021-01-10:
I'll add two settings in the printer tab.
It's mostly just rounding for printing the number in the file.

> What would be the downside to the finer resolution?

some firmware can't parse a number that has too many decimals.

chriskoups, 2021-01-10:
Thank you! As I said, I don't know how much it may improve things but it will not harm and I think is an easy addition...

closed 2021-06-26 by supermerill