supermerill/SuperSlicer #853: How do I use the "milling cutter" feature?

AmyTheCute, 2021-01-05:
### Version
2.3.55

### Operating system type + version
windows 10
### 3D printer brand / version + firmware version (if known)
Doot Changer 2.0 Klipper 0.9.1

I see that there is a "milling cutter" option now. if I choose 1 i can only see a setting for diameter, gcode and z hop. how do I actually use it? I have a tool changing printer and I planned on adding a mill to it!

Thanks

supermerill, 2021-01-10:
In the output tab, you can, there is a section to activate it, with some properties.
It basically runs over the perimeters.
I'm sure there are some oversights I made, as I can't test it IRL.
So post here the problems you have using it and possible improvements you can think about.

AmyTheCute, 2021-01-10:
> In the output tab, you can, there is a section to activate it, with some properties.
> It basically runs over the perimeters.
> I'm sure there are some oversights I made, as I can't test it IRL.
> So post here the problems you have using it and possible improvements you can think about.

what/where is the output tab? I can't find a tab named output in my slicer!

supermerill, 2021-03-18:
> pretty scared to try it tbh!

look at the gcode first, then try it without any cutter, then with a cutter.

closed 2021-06-04 by supermerill

adamrumjahn, 2022-04-28:
Does this assume that the milling cutter/tool has the same XYZ location as the printing nozzle?  Or is there somewhere where this should be configured?

I didn't see any actual output in the gcode that suggests that it turns on a milling tool and cuts.  It this workflow meant purely for increasing the size of the print so that there is material to remove? or is there actual cutting meant to be in the g-code as well?