supermerill/SuperSlicer #854: Crashes to desktop (no errors)

Makhaira, 2021-01-05:
### Version
2.2.54

### Operating system type + version
Win10

### Behavior
SS will frequently crash immediately on launch - the program appears, freezes for a few seconds, and then exists to desktop with no message.
I am also getting some crashes when switching between printer profiles. This is regardless of whether STLs are already loaded or not.

Is there some sort of crash log I could post to help?

Kveri, 2021-01-06:
I can confirm, past few weeks I also get random crashes, usually once a day, nothing serious if you use ctrl+s a lot, but still. No error, just straight quit.

supermerill, 2021-01-10:
No crash log i known.
If you can reproduce the crash, tell me how.

Makhaira, 2021-01-10:
Seems to be having a picture assigned to the bed from what I can deduce so
far

Kveri, 2021-01-12:
> No crash log.
> If you can reproduce the crash, tell me how.

is it possible to somehow enable crashlog or crashdump? I get totally random crashes but only once every two days. I didn't spot any correlation to what I do, which STLs I use, etc.

supermerill, 2022-02-03:
I fixed many bugs that can crash the slicer, so hopefully the current stable shoudln't crash anymore.

> is it possible to somehow enable crashlog or crashdump?

only if built with debug flags. I can give you one.

LucaRood, 2022-02-06:
I'm having the same issue (or at least the same symptom), on Linux. The latest official release build crashes immediately upon launch, but not consistently (perhaps about 50% of the time, or a bit more).

I'm not sure if it's the same issue and/or if it's platform specific, but since it's the same symptom, I figured it'd make sense to reply here.

Since the official build seems to conveniently include debug symbols, I was able to generate the backtrace below. I haven't had time to get the master branch building to look into it myself, but I hope this helps with finding the issue.

<details><summary>Backtrace</summary>

```
#0  0x000055555610aa21 in Slic3r::GUI::GLTexture::load_from_svg_files_as_sprites_array(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::pair<int, bool>, std::allocator<std::pair<int, bool> > > const&, unsigned int, bool, unsigned int) ()
#1  0x00005555560c8049 in Slic3r::GUI::GLGizmosManager::generate_icons_texture() const ()
#2  0x00005555560c82ed in Slic3r::GUI::GLGizmosManager::render_overlay() const ()
#3  0x00005555560a0a50 in Slic3r::GUI::GLCanvas3D::_render_overlays() const ()
#4  0x00005555560a1eb5 in Slic3r::GUI::GLCanvas3D::render() ()
#5  0x00005555560a3934 in Slic3r::GUI::GLCanvas3D::_refresh_if_shown_on_screen() ()
#6  0x00005555560a3a90 in Slic3r::GUI::GLCanvas3D::on_set_focus(wxFocusEvent&) ()
#7  0x0000555556d07159 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const (this=0x555557b62630, handler=0x5555586eba50, functor=warning: RTTI symbol not found for class 'wxEventFunctorMethod<wxEventTypeTag<wxFocusEvent>, Slic3r::GUI::GLCanvas3D, wxFocusEvent, Slic3r::GUI::GLCanvas3D>'
..., event=...)
    at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:671
#8  0x0000555556da7550 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=..., handler=0x5555586eba50, event=...)
    at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1415
#9  0x0000555556da8219 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1887
#10 0x0000555556da7996 in wxEvtHandler::TryHereOnly(wxEvent&) (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1608
#11 0x0000555556da8db7 in wxEvtHandler::TryBeforeAndHere(wxEvent&) (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:3912
#12 0x0000555556da77e5 in wxEvtHandler::ProcessEventLocally(wxEvent&) (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1545
#13 0x0000555556da777c in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1518
#14 0x0000555556da7a91 in wxEvtHandler::SafelyProcessEvent(wxEvent&) (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1636
#15 0x0000555556badb52 in wxWindowBase::HandleWindowEvent(wxEvent&) const (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/wincmn.cpp:1555
#16 0x0000555556c3f79b in wxWindow::GTKProcessEvent(wxEvent&) const (this=0x5555586eba50, event=...) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/window.cpp:1536
#17 0x0000555556c47ea3 in wxWindow::GTKHandleFocusIn() (this=0x5555586eba50) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/window.cpp:4444
#18 0x0000555556c40efb in gtk_window_focus_in_callback(GtkWidget*, GdkEventFocus*, wxWindow*) (win=0x5555586eba50) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/window.cpp:2048
#19 0x00007ffff7418f38 in  () at /usr/lib/libgtk-3.so.0
#20 0x00007ffff6eb6d8f in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#21 0x00007ffff6ed2718 in  () at /usr/lib/libgobject-2.0.so.0
#22 0x00007ffff6ed340b in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#23 0x00007ffff6ed4330 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#24 0x00007ffff76cfc45 in  () at /usr/lib/libgtk-3.so.0
#25 0x00007ffff76cba47 in gtk_widget_send_focus_change () at /usr/lib/libgtk-3.so.0
#26 0x00007ffff76de8d7 in  () at /usr/lib/libgtk-3.so.0
#27 0x00007ffff76def2a in  () at /usr/lib/libgtk-3.so.0
#28 0x00007ffff6eb57eb in g_cclosure_marshal_VOID__OBJECTv () at /usr/lib/libgobject-2.0.so.0
#29 0x00007ffff6ed41c0 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#30 0x00007ffff6ed4330 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#31 0x00007ffff6ed41c0 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#32 0x00007ffff6ed4330 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
[... cut ...]
#65 0x0000555556d588cf in wxEntry(int&, wchar_t**) (argc=@0x55555799b8b0: 1, argv=0x555557b6fc80) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:507
#66 0x0000555556d589a5 in wxEntry(int&, char**) (argc=@0x7fffffff6600: 1, argv=0x7fffffffe228) at /home/runner/work/SuperSlicer/SuperSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:519
#67 0x0000555555d7c342 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#68 0x0000555555751953 in Slic3r::CLI::run(int, char**) ()
#69 0x000055555572bb7a in main ()
```
</details>

supermerill, 2022-02-10:
Thank you very much for your debugging investigation.

For the fix, I propose to change for
```
    if (is_limited && int_val < 50)
        int_val = 50;
    else if (int_val < 1) {
        int_val = 10; //buggy code? issue supermerill/superslicer#854
        assert(false);
    }
```
`auto_toolbar_size` doesn't seem to be set by anything and isn't used anywhere else... Maybe prusa needs it for something (it was added in may 2020) but forgot about it?

For the `ENABLE_RETINA_GL` vs `__APPLE__` These kinds of GUI things are extremely time-consuming. I let prusa investigate & fix them. If you can test gtk3 with & without and find the diff, you may be able to fix it for them.

LucaRood, 2022-02-10:
While I agree that your proposed code mitigates the issue, I wonder if this is a wise decision, as it just hides the actual issue that is further up. I feel like other unexpected issues could easily crop up from this. For instance, if this setting happens to be used anywhere else, the values won't match. And besides, if this setting is somehow being changed somewhere, it is also plausible to imagine that the bug might affect other settings as well.

Of course, it's not up to me, and I'm very new to this codebase, but I'd advise against this sort of band-aid. Though I do admit that it's good to have a safeguard (such as what you wrote), in case users set the setting itself to an implausible value. But in this case, it'd be good to also find where the actual issue is coming from.

As for the retina handling, yeah, I figured it's mostly a job for upstream, as it is indeed a lot of very dense code, but I figured I'd bring it up just in case. But it's not really important.

supermerill, 2022-02-10:
That's why I added the assert, so it can trigger if it's on debug mode.
Let's be real, I can't replicate the issue., and I won't spend days hunting it. So until someone (or prusa) comes with a better solution... it's better than nothing.
Does it work on your computer if you compile with the fix?

LucaRood, 2022-02-13:
Had some time to debug further, and come up with a proper fix (though, to answer your question, yes, your change did stop the crash from happening). I've submitted the fix at #2368, which addresses the source of the negative value.

This problem seems to be caused by the canvas not being fully initialised when icon scaling starts being applied. Though I tested on another system and didn't hit the crash, so it seems to be dependent on which WM is used.

With any luck this is the same crash as in the original report, though at this point it's looking unlikely. If this crash affected Windows, I'd expect it to crash for everyone on Windows (as it's actually config independent), and since that's not the case, it's probably another issue.