supermerill/SuperSlicer #857: First Layer Ignores "Only retract when crossing perimeters"

StrikeEagleCC, 2021-01-06:
### Version
2.2.53 &2.3.55
### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Modified Ender 3 + Klipper

### Behavior
"Print Settings > Infill > Advanced > Only retract when crossing perimeters" ignored on first layer

* _Steps needed to reproduce the problem_
Start with a blank plate. Add shape > box. Rotate about z axis 30 degrees. enable "Only retract when crossing perimeters". Set "Minimum travel after retraction" to 0.5" Slice and observe first layer.

 * _If this is a command-line slicing issue, include the options used_
N/A

* _Expected Results_
Retractions only occur when the subsequent travel move crosses an open perimeter

* _Actual Results_
On the first layer only, retractions occur prior to every travel move

  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]

#### Project File (.3MF) where problem occurs
[First Layer Retractions Example - SuperSlicer_2.3.55.zip](https://github.com/supermerill/SuperSlicer/files/5773370/First.Layer.Retractions.Example.-.SuperSlicer_2.3.55.zip)

If this behavior is intended, I think an tick box to toggle it would be great. Also, I'm assuming the travel moves where the rectilinear solid infill meets the perimeters at a steep angle is intended behavior, to prevent extrusions from overlapping themselves.

Thanks for your great work!

supermerill, 2021-01-10:
There is a safety thing for the first layers, to be able to z-hop higher. Maybe that's an unintended consequence.

supermerill, 2022-02-03:
seems fixed. Sorry for late update.

closed 2022-02-13 by supermerill