supermerill/SuperSlicer #860: Variable layer height appears inverted

ProfEngr, 2021-01-06:
### Version
Last alpha from Dec '20

### Operating system type + version
Win 7

### Behavior
Enabled the feature. Clicked Adaptive then Smooth. Details are showing high layer heights where vertical, low-detail walls are showing minimum layer height.

[image]

theophile, 2021-01-07:
That looks right to me. What are you expecting to see?

ProfEngr, 2021-01-07:
Areas that have filets should be at 0.12 while featureless walls should be 0.3.

theophile, 2021-01-07:
It's hard to see the problem without being able to see the rest of the model. What did it look like before you smoothed it?

limited660, 2021-01-07:
@theophile the area in green doesn't need low layer height because it's straight, the area in orange needs it because of the fillets.

theophile, 2021-01-08:
Something on the other side of the model presumably has more details in the green area. It looks like it tried to use a lower layer height for the fillets but it was smoothed out. Try it again without smoothing.

supermerill, 2021-01-10:
I'll look into it when i'll have some more time and less serious bugs.

ProfEngr, 2021-01-12:
> Something on the other side of the model presumably has more details in the green area. It looks like it tried to use a lower layer height for the fillets but it was smoothed out. Try it again without smoothing.

All 4 corners are the same geometry. It's a simple box for a NodeMCU. I'll try some other parts and see if it was just a fluke.

ProfEngr, 2021-01-12:
I think it might just be that specific geometry. I tried a few other STLs and the algorithm seems to be ok. Sorry to bother everyone.

closed 2021-01-12 by ProfEngr