supermerill/SuperSlicer #861: Is this a bug? Ironing on every raft interface?

scott0122, 2021-01-07:
### Version
Latest Master (2.3.55)

### Operating system type + version
Windows 10 x64

### Behavior
Setting: 
 Raft layers:3
 Interface layer:3
 Interface pattern: Ironing

Layer 1(raft interface 1) is ironing
[image]

Layer 2(raft interface 2) is ironing
[image]

Layer 3(raft interface 3) is ironing
[image]

supermerill, 2021-01-10:
did you select ironing as interface pattern? IIRC, it's the pattern used for the raft (there is already an issue to change that)

scott0122, 2021-01-10:
@supermerill Yes, interface pattern is Ironing. Should it be only ironing on top raft layer?
Setting:
Raft layers:3
Interface layer:3
Interface pattern: Ironing

supermerill, 2021-01-10:
I have to create more pattern settings to decouple things.

closed 2021-06-04 by supermerill