supermerill/SuperSlicer #862: Setting temperatures for RepRap firmware

Phil1988, 2021-01-07:
### Version
2.2.53

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
A Duet Wifi with RRF2

### Behavior
The print starts with homing and then the print head moves directly to the starting position and goes around the shapes without extruding something.

The gcode ist looking like this:
`M107
G10 P1 R200 S195 ; sets the standby temperature
G10 P1 S200 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
G10 P1 S200 ; set temperature
M116 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
T1
; Filament gcode
;BEFORE_LAYER_CHANGE
;0.24`

I am wondering about the G10 for setting the temperature as well as for the M116 command.

I would expect something with M104 and M109 like:
`M107
M104 S205 T1 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
M109 S205 T1 ; set temperature and wait for it to be reached`

I am also getting the error 
"Warning: Tool 1 was not driven because its heater temperatures were not high enough or it has a heater fault"

I am having 2 printheads.

neophyl, 2021-01-07:
G10 is a valid temp setting code for reprap according to 
https://reprap.org/wiki/G-code#G10:_Set_tool_Offset_and.2For_workplace_coordinates_and.2For_tool_temperatures

might want to read https://github.com/supermerill/SuperSlicer/issues/451 as the change to G10 for reprap is mentioned in that issue too.

Phil1988, 2021-01-07:
Thanks @neophyl 

I read that but I dont understand, why the doku says a G10 + M116 will do the same as a M109 would normally do, but it does not work for me.

Do you have an idea?

I know I can do a workaround with 
`M109 S[first_layer_temperature] ; wait for extruder final temp`
but I dont know how I can enter the right tool (T0 or T1) and I dont even understand why this workaround is necessary. (Because RRF should support the G10 + M116...)

neophyl, 2021-01-07:
Sorry I dont have a duet or anything with reprap in.  Both of my printers run versions of Marlin.  

I just remember reading/participating in that thread so seeing you mention the G10 jogged the old brain cells.

supermerill, 2021-01-10:
if you want the old behavior, you should select "sprinter" as firmware

Phil1988, 2021-01-14:
So I got an answer from the folks at the duet forum.

The **M116 waits** for the temps reached for the **printbed** and the **active tool**.
As you can see in the code that superslicer produces, it does make the tool active (T1) **after** the wait command (M116). 

So it does not wait... ;)

It should be changed, so that it generates a GCODE where the active tool (T0, T1 ...) comes before M116 to work correctly with reprap firmware.
In my opinion it does not work as intended and I would appreciate, if you can change the sequence of the commands accordingly :)

supermerill, 2021-01-23:
I'll move it after the tool selection.
Most safe way is to write all of that in your start gcode.

Phil1988, 2021-01-23:
Thanks supermerill!

I would add this to my start gcode, but I don't know how to insert a "placeholder" for the tool (I am using 2 tools regularly) :)

closed 2021-06-04 by supermerill