supermerill/SuperSlicer #864: Variable layer height only takes settings from tool 0

Phil1988, 2021-01-07:
### Version
2.2.53

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
A Duet Wifi with RRF2

### Behavior
When I use the variable layer height feature, the object just takes the values for min and max height of tool 0.
Also if the part/model is selected to be printed with tool 1.

It would be great if it uses the variable layer heigt values for the set tool of that part :)

#### Project File (.3MF) where problem occurs
[adaptive_layer_for_tool1.zip](https://github.com/supermerill/SuperSlicer/files/5782531/adaptive_layer_for_tool1.zip)

supermerill, 2021-01-10:
May be a better idea, yes. I'll see if the information is where it's needed.

If the object is printed with multiple tool, I'll see if I can get the new min-max.

Phil1988, 2021-01-11:
In my case, the part is only printed by one tool (the second nozzle - so tool #1 (counting from 0)).
And for me it would be "sufficient" it the active tool is used for calculating the variable layer height.

I think there might be also cases where users want multiple tools to print one piece with the variable layer feature, but this would be even more rare to find :)

Thanks for taking a look into that supermerill :)