supermerill/SuperSlicer #866: Only one perimeter broken since commit

Stephan3, 2021-01-09:
### Version
latest nightly

### Operating system type + version
win64

### 3D printer brand / version + firmware version (if known)
vorons

### Behavior
superslicers continiues single perimeter on some parts after detecting the first occurence of a toplayer, see screenshot. Increasing min width top surface to values like 10000% makes it go away. its like simply deactivating it. I reverted that commit which made it working again. 
https://github.com/supermerill/SuperSlicer/commit/525d23d02cff2a8bc0d40efcde94ef9d68f63ba6

[image]

#### Project File (.3MF) where problem occurs
[supertest.zip](https://github.com/supermerill/SuperSlicer/files/5790417/supertest.zip)

neophyl, 2021-01-09:
You have Only one perimeters on top layers enabled in Print Settings>Perimeters & Shell>Quality.
Doing only a single perimeter on top layers to allow more space for infill is exactly what that setting is designed to do.

Stephan3, 2021-01-09:
yes, you missed that we are in half of the heights. the postition is "midpart"

supermerill, 2021-01-10:
same issues as #745 #733

closed 2021-01-24 by Stephan3