supermerill/SuperSlicer #868: Gap Fill Very Inconsistent

CCS86, 2021-01-09:
Win10 x64 / 2.3.55.2

I am having a very hard time getting gap filling to act consistently.

Some areas fill pretty well (green), other areas start to gap fill, stop too early (yellow), some areas don't even try (red), and one area actually had room for one more actual perimeter, but gap filled partially instead (purple).

[image]

[Servo Arm1.6.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5791690/Servo.Arm1.6.3mf.txt)

supermerill, 2021-01-10:
Some may be evicted because they are too small, did you try with a very low min area?

Some are more problematic to resolve/display. On the bottom part, you put an arrow on an empty surface, but it's empty because of the way the yellow loop is draw, as it should be filling the are for a width/2 more, like on the right side (wich print fine, because it's making a nice curve vs steep direction change). I'll have to print it with the real polygon to see if there is any missing areas, and by how much.

CCS86, 2021-01-11:
Yes, I have the gap filling threshold to 30% (of a 0.4mm nozzle), which is quite a bit smaller than default. 

Let me know if I can provide anything else to help.

supermerill, 2021-01-23:
An exemple of "visual artifact"
[image]

90% of your issue is that: just that the tip isn't showed / printed.

supermerill, 2021-01-23:
It's maybe possible to change the way it's sliced to forbid angles sharper than 120°, but it's a big task and I really don't have the time for that.

closed 2021-06-14 by supermerill