supermerill/SuperSlicer #869: Strange artifacts on curves and misaligned perimeters

uepsie, 2021-01-10:
### Version
2.3.55.4

### Operating system type + version
Windows 10 64Bit Build 19042.685

### 3D printer brand / version + firmware version (if known)
Prusa MK3 + Klipper

### Behavior
When slicing a benchy with default parameters for Prusa MK3 (0.4 nozzle, Prusament PLA, 0.2 Quality Profile) there are strange artifacts also reflected in the print.

[image]
[image]

#### Project File (.3MF) where problem occurs
added the zipped 3mf:
[#3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/5793278/3DBenchy.zip)

uepsie, 2021-01-18:
I just tried the same model with same settings in 2.3.55.1

[image]

[image]

It seems to work flawless here

supermerill, 2021-01-18:
seems like the printer -> general -> min extrusion length is too high,  it's better with a value of 0.001

uepsie, 2021-01-18:
interestingly this fixed it.. so is it a problem with Prusa's default settings?

uepsie, 2021-01-19:
Fixed by setting min extrusion length to 0.001
Closing this one

closed 2021-01-19 by uepsie