supermerill/SuperSlicer #870: Internal Error: boost::bad_any_cast

uepsie, 2021-01-10:
### Version
2.3.55.4

### Operating system type + version
Windows 10 64Bit Build 19042.685

### 3D printer brand / version + firmware version (if known)
Prusa MK3 + Klipper

### Behavior
SuperSlicer crashes on click on "Printer Settings" Tab (even after fresh start)
[image]
[image]

Pavulon87, 2021-01-11:
Same here. Slicing works. "Printer settings" crash.

Pavulon87, 2021-01-11:
Quick fix:
from **resources/ui_layout/printer_fff.ui**
remove: 
```
	line:Size for Gcode
		setting:id$0:label$Small:thumbnails
		setting:id$1:label$Big:thumbnails
	end_line
```

uepsie, 2021-01-11:
I don't have those lines in my printer_fff.ui

Pavulon87, 2021-01-11:
> I don't have those lines in my printer_fff.ui
Is your file different than https://github.com/supermerill/SuperSlicer/blob/master/resources/ui_layout/printer_fff.ui ?

uepsie, 2021-01-12:
Oh - my bad. Typo in my search
indeed, removing those lines help

khumarahn, 2021-01-17:
Same here, on arch linux.

supermerill, 2021-01-18:
fixed yesterday

closed 2021-06-04 by supermerill