supermerill/SuperSlicer #872: Hebrew Translation

SM-26, 2021-01-11:
### Version
2.3.55.2

### Operating system type + version
Windows 10, 64Bit eng.

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Behavior
I'd like to help out with Hebrew Translation.
I've read the [Wiki](https://github.com/supermerill/SuperSlicer/wiki/Creating---updating-a-translation) but still I don't see my new language

#### Project File (.3MF) where problem occurs
I've created a new Fork, with my new branch.
Here is [PR 871](https://github.com/supermerill/SuperSlicer/pull/871)

But I don't understand what I'm missing in order to lunch SC with my (new) language.
I just need a bit more help with getting the ball rolling about adding a language.

Please help me out help others out.
Thanks in advance.

SM-26, 2021-01-13:
[image]
I'm getting there, but still having some problems.
Fonts? encoding?

supermerill, 2021-01-18:
I created a he_IL directory in the resources/localization folder.
I copied into it a SuperSlicer.mo from another directory.
It works as intended, proposing the Hebrew entry and translating the strings as set in the .mo

SM-26, 2021-01-18:
he_IL
not just he
That was the piece I was missing!
thanks.

I'll keep working on my translation, huge thanks for SuperSlicer.

SM-26, 2021-01-21:
Ok, I think I need a bit more help.
Now hebrew is an option, but it does not change anything. I don't see any change that I've made.

I've used OmegaT for translation, but now I've found out it really messes with the file internal structure and render it unusable.
Poedit don't like how the TODO file is looking and would not accept it.

Please don't tell me notepad++ is the only option.

supermerill, 2021-01-23:
I don't have any problem with poedit:
[image]

Maybe setting windows to use utf-8 will help: https://stackoverflow.com/questions/56419639/what-does-beta-use-unicode-utf-8-for-worldwide-language-support-actually-do

SM-26, 2021-01-25:
So maybe my settings.ini is off when I build it?
here is how it looks:

``` ini
data = C:/myDir/localization/en/PrusaSlicer_en.po
data = C:/myDir/localization/en_US/Slic3r++_en.po
data = C:/myDir/localization/en_US/PrusaSlicer_en.po

input = C:/myDir/localization/SuperSlicer.pot
todo = C:/myDir/localization/TODO.po
output = C:/myDir/localization/he_IL/SuperSlicer.po
```

and the TODO.po comes out like this:
[image]

I did some digging, looks like Poedit doesn't like the header.
so I copy the header from the english po file and modified by hand.
```
# SOME DESCRIPTIVE TITLE.
# Copyright (C) YEAR THE PACKAGE'S COPYRIGHT HOLDER
# This file is distributed under the same license as the PACKAGE package.
# FIRST AUTHOR <EMAIL@ADDRESS>, YEAR.
#
msgid ""
msgstr ""
"Project-Id-Version: \n"
"Report-Msgid-Bugs-To: \n"
"POT-Creation-Date: 2019-04-09 14:34+0200\n"
"PO-Revision-Date: 2019-04-09 15:04+0200\n"
"Language: he_IL\n"
"MIME-Version: 1.0\n"
"Content-Type: text/plain; charset=UTF-8\n"
"Content-Transfer-Encoding: 8bit\n"
"Last-Translator: Oleksandra Iushchenko <yusanka@gmail.com>\n"
"Language-Team: \n"
"X-Generator: Poedit 2.0.8\n"
```

but still the same error.

I've also tried to open the POT file in poedit and from that ask him to build a po file.
and after renaming it to SuperSlicer.po (and making a mo for it)
I could not see any changes of the UI from my translations.

Do you have some kind of a "blank" po file with everything (from SuperSlicer & PrusaSlicer & Slic3r ) that you know works well?
Thanks in advance

supermerill, 2021-03-19:
> Do you have some kind of a "blank" po file with everything (from SuperSlicer & PrusaSlicer & Slic3r ) that you know works well?
Thanks in advance

That's something I have to work on, even my current tools doesn't catch everything. I'll post a message here when I'll make some progress

SM-26, 2021-06-07:
could you please keep this open?
I'm seeing all the other people that are adding translations and I'm a bit envies.

supermerill, 2021-09-06:
sorry, i closed all "question" issues times to times. I'll change the tag.