supermerill/SuperSlicer #873: Internal error: invalid map<K> key

SimonSolar2C, 2021-01-11:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
2.3.55

### Operating system type + version
_What OS are you using (and version)_
Win10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* Crash
* _Steps needed to reproduce the problem_
* Changing printer profiles with unsaved changes
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

uepsie, 2021-01-11:
same here with 2.3.55.4

Kveri, 2021-01-12:
same here, I get these randomly, still haven't found when and why. Some debug output would be useful.

rveda, 2021-01-12:
I got the same error when trying to change printer profiles

supermerill, 2021-01-18:
> Changing printer profiles with unsaved changes

can you propose a bit more detailed procedure?
can't make it crash.

supermerill, 2021-06-26:
should be fixed by now

closed 2021-06-26 by supermerill