supermerill/SuperSlicer #875: M109 and M190 Injected after PRINT_START macro

MarcPot, 2021-01-13:
### Version
2.3.55.2

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Klipper

### Behavior
When using the Klipper gcode flavour there is still a M109 and M190 after my PRINT_START macro.
This causes the printer to wait a sec or 2 after my print start (purge line), causing it to ooze before starting the actual print.

The reason for the printer to wait is because it's not always exactly within the M109 range.

A solution would be an (expert mode) checkbox to disable everything between 
`; custom gcode end: start_gcode` and `;LAYER_CHANGE`

#### Project File (.3MF) where problem occurs
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/5809133/Shape-Box.zip)

ProfEngr, 2021-01-20:
I use this as my printer tab custom g-code:

```
; process STARTGCODE
M107
M190 S[bed_temperature] ; just in case the slicer doesn't do it (i.e. 2.2.53)
START_PRINT_PLA ; Klipper macro
M104 S[first_layer_temperature_0] ; finish heating E0 during purge and skirt
PRIME_LINE ; Klipper macro. Must call above items first
; process PARTS
```
Here is my Klipper macro:
```
[gcode_macro START_PRINT_PLA]
gcode:
    CLEAR_PAUSE
    set_gcode_offset z=0.000      # higher offset = less squish. added to z_offset below
    M107 ; stop part fan
    M220 S100 ; reset feedrate
    M104 S180 ; start to heat E0, oozing above 185
    G21 ; set units to millimeters
    G90 ; use absolute coordinates
    G28 ; home all axes
    ; Klipper auto-loads the default mesh at the end of this file (defunct)
    BED_MESH_PROFILE LOAD=MESH{printer.heater_bed.target|int}C ; PLA mesh stored end of file
    M82 ; use absolute extruder distances
    G92 E0.0 ; zero the extruder position
    G1 X10 Y10 Z15 F6000 ; ooze here while heating
    M109 S190 ; wait for E0 to preheat
    SET_RETRACTION UNRETRACT_EXTRA_LENGTH=0 RETRACT_LENGTH=0.35
    ; slicer sets first layer hotend temp next
```

I also use Klipper as the firmware flavor w/ machine limits disabled.

MarcPot, 2021-01-20:
Hey @ProfEngr,

Thanks for wanting to help, but I think you misunderstood my issue. Because if I’d use what you have there I’d still have the same problem. Because there would still be a M109 and M190 after the PRIME_LINE.

ProfEngr, 2021-01-20:
I had something similar to this a while back. That's why I have the bed and nozzle temp variables specifically called out in the custom section. It was an issue w/ the firmware flavor I was using at the time. I just left it in because it kept working.

aka13-404, 2021-02-26:
I would like to confirm, that I get unwanted m104, 109 and 190 injected in the beginning of the gcode. I have them processed entirely on the klipper side, and do not want them in my gcode. AFAIK there is still no way to disable them?
[image]

randellhodges, 2021-02-27:
I wonder if these lines: https://github.com/supermerill/SuperSlicer/blob/201c6120b6c6122843ea04a388c3d93eb9cce1b7/src/libslic3r/GCode.cpp#L1406-L1410 need a similar gcode_flavor check like these lines: https://github.com/supermerill/SuperSlicer/blob/201c6120b6c6122843ea04a388c3d93eb9cce1b7/src/libslic3r/GCode.cpp#L1320-L1328

Frix-x, 2021-04-14:
I confirm, I got the same unwanted behavior.

This come from PrusaSlicer -> if you don't put explicitly M109/M190 into the custom start gcode of the slicer, it will add it for you as a security for Marlin printers. This is a regression in SuperSlicer coming from the last PrusaSlicer merge I think.

supermerill, 2021-04-29:
I'm going to add a settign to force disable auto-added heating lines.

MarcPot, 2021-04-29:
Thanks Merill! I'll close when I get to testing it :)

MarcPot, 2021-05-03:
@supermerill Sadly gotta report that the function is currently broken in the [C/C++ Nigthly windows x64 C/C++ Nigthly windows x64 #296. ](https://github.com/supermerill/SuperSlicer/actions/runs/804665432)

Sliced with the function off:
[image]

Sliced with the function on:
[image]

3mf:
[raspberrypi_mounting_bracket_x1_rev1.zip](https://github.com/supermerill/SuperSlicer/files/6413259/raspberrypi_mounting_bracket_x1_rev1.zip)

supermerill, 2021-05-03:
@MarcPot everyhting is fine.
the difference are:

```
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M104 S245 ; set temperature
M109 S245 ; set temperature and wait for it to be reached
M190 S85 ; set bed temperature and wait for it to be reached
```

are not written with the 'only custom gcode' so be sure to have them in your start gcode!

also there is a M104 & 140 a bit alter in the print with the option activated, but it's nothing to be worry about, imo.

MarcPot, 2021-05-03:
You're right, should've looked at the files instead of the viewers.
Thanks for the great work Merill!

Closing this issue since it has been resolved and the bug is unrelated to the issue.

closed 2021-05-03 by MarcPot