supermerill/SuperSlicer #876: Offering my translation service for free

beistline, 2021-01-13:
Hello, as the title says: I have a degree in Translation and ME. I love the software and want to help Translate it to European Portuguese. Let me know if you are interested. Thank you

supermerill, 2021-01-23:
noted.
I have to streamline the process a bit, as I forgot to include some files.
Will go back to you here when it's ready.

beistline, 2021-02-05:
Can't wait.

beistline, 2021-11-11:
Any updates on this?

supermerill, 2021-11-12:
yes, sorry.
I updated the doc available [here](https://github.com/supermerill/SuperSlicer/blob/master/resources/localization/README.md)
Now, using the python tool, it will generate you a list of text to translate, and may propose existing translations that are similar enough.
If you have any problem understanding or following the doc, tell me so i can help you & update it.