supermerill/SuperSlicer #877: [Bug] Crash and Unhandled unknown exception; terminating the application.

digitalentropy, 2021-01-13:
### Version
2.3.55.4

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2.5S

### Behavior
After importing a config bundle from SS 2.2, SuperSlicer-alpha produces an "Unhandled unknown exception" error when attempting to view the Printer settings tab.

Two errors are observed when this happens:

First Error: Unhandled unknown exception; terminating the application.
Second Error: Internal error: boost::bad_any_cast: failed conversion using boost::any_cast
[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/5810511/SuperSlicer_config_bundle.zip)

uepsie, 2021-01-13:
Have you checked #870?

digitalentropy, 2021-01-14:
> 
> 
> Have you checked #870?

Ah, somehow that one did not come up in my search when I was looking through open issues for error messages.

This bug appears to be a dupe of that one. Closing.

closed 2021-01-14 by digitalentropy