supermerill/SuperSlicer #878: Notified of new release - New release checking is off, and NO new release...

TheAndr0id, 2021-01-13:
### Version
2.2.53.2

### Operating system type + version
Linux x86_64

### Behavior
This may very well be fixed in 2.3.x. - I thought it worth bringing up.

I just received a "new update available" popup when starting SuperSlicer.  I have "Check for application updates" unchecked in the application Preferences, and after checking the releases page here, the last release was in 2020.

Prusa on the other hand has officially released 2.3.0.  I'm guessing some sort of hold over from the Prusa fork still has SuperSlicer checking for new releases from Prusa.

I didn't grab a screenshot, as I didn't think this was an issue.  Now that I think more I believe the message stated "2.3.0" was available which doesn't match SuperSlicer's versioning.

So, two issues here,

1. The update check shouldn't be happening as it was turned off in preferences.
2. It's checking for the wrong updates (Looking at PrusaSlicer not SuperSlicer).

jariivanoff, 2021-01-19:
And clicking on the "Go to download page" gets you to Prusas homepage.

supermerill, 2021-01-23:
already fixed in the next version, sorry for that.

closed 2021-06-04 by supermerill